homieiot/homie-esp8266 #2: MQTT reconnection delay

furyfire, 2016-01-14:
Tiny bug in your reconnection routine for MQTT
You reuse
_last_wifi_reconnect_attempt
instead of the correct
_last_mqtt_reconnect_attempt

Fixed in my fork

marvinroger, 2016-01-14:
You're right, thanks for spotting it. Is there anything you don't like/understand in the API?

closed 2016-01-14 by marvinroger

furyfire, 2016-01-18:
I have not used the HomieServer - I instead run my own MQTT broker for handling my data.
One thing that did pop up was that data for a given device is stored under /devices/[name]/ I think that too easily could result in trouble. I suggest the device name is moved into a structure similar to localip and firmware version
/devices/[chipid]/$name 
This way a user won't set up two devices both called "tempsensor" and have them conflict with each other on the server

commit 6097d902 (2016-01-14): Fix #2 - thanks @furyfire for spotting