homieiot/homie-esp8266 #3: Add config for MQTT port

enavarro222, 2016-01-16:
To use it with a std MQTT broker it could be nice to make MQTT broker port configurable.

It's straightforward to add such option, however in a "pure" Homie framework it's just an other useless option... 

What do you think to add an `homie_port` option that take the value from the `HOMIE_PORT` constant if not given in the config JSON ?

I can propose a PR in this sense.

marvinroger, 2016-01-16:
I think this is a good idea, but the fact is an Homie server is actually 2 servers: the MQTT broker on 35589 and the OTA HTTP server on port 35590, that respond on the `/ota` path. So maybe you could also add an `homie_ota_port` and an `homie_ota_path`, so that a developer could implement the OTA server on its own HTTP server?

What do you think about it? If you have another ideas, even breaking ones, go ahead! Homie is still young and, well, not used by a lot of people so there will probably be a lot of breaking changes before reaching 1.0.0. ;)

enavarro222, 2016-01-16:
Nice. I'll do that.
It's just a small change.

A "breaking" idea could be to have a more flexible option system. With the current options by default + other that a dev can declare in his setup for a particular device. But it might be a wrong way as such option can be given by MQTT...

marvinroger, 2016-01-17:
What do you mean by options?

PR #7 (2016-01-26): Add MQTT port and OTA path & port to config
This should close #3, this also include #6 but with to call to the log mth : 
* when config is saved in config mode
* at setup time in normal mode
closed 2016-01-26 by marvinroger