homieiot/homie-esp8266 #5: Strange "wdt reset"

enavarro222, 2016-01-17:
Hi, 

I'm testing it with a nodeMCU 0.9, I got the following log:

Boot in config mode, ok. Then I send json config and :
```
Received config request
✔ Configured

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

I reflash the FW, it boot in normal mode.
Then when I manually ask for a reboot in config mode:
```
** Booting in normal mode **
Attempting to connect to WiFi
Attempting to connect to MQTT
Sending initial informations
Temperature: 22.00 °C
Resetting
↻ Rebooting in config mode

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

I use mosquito as MQTT broker.

Maybe some `delay()` should be added to handle wifi processing ? or `mqtt.loop()` ?

marvinroger, 2016-01-17:
The issue comes from the Arduino for ESP8266 (see https://github.com/esp8266/Arduino/issues/1017)
Basically, an `ESP.restart()` after a flash triggers a wdt reset. 

To confirm this is the issue, try this:

1. Flash the firmware
2. Manually reset the board
3. Repeat the config request/reset

You will see that the board is actually rebooting.

marvinroger, 2016-01-17:
No that when stuck on a wdt reset, you don't even have to reflash the firmware: just reset the board manually and it will boot in the correct mode.

enavarro222, 2016-01-17:
Indeed, this doesn't appear when I hard reset power of the board after flashing !

We can leave this ticket open for documentation, until bug isn't fix in Arduino/ESP8266.

Resume: With NodeMCU 0.9 (maybe 1?), you need to manually power reset the board after flashing to have reboot working.

enavarro222, 2016-01-17:
I'm not familiar with nodeNCU, Is there a way to reset it without plug/unplug it ?

marvinroger, 2016-01-17:
Yes, let's leave it open for reference. It is not only a problem with NodeMCU, it is for every ESP8266 ;)

On NodeMCU 1.0 there is a RST button. It seems like NodeMCU 0.9 doesn't have such a button, so you'll have to power reset it!

enavarro222, 2016-01-17:
Ok, thanks for the quick answer !

Dylantje, 2016-01-19:
Dear alll...
I use a Nodemcu v.09 and v.10 i did have have work my esp correct.
after a new flash i did have a problem with the esp...
<code> 
wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
ªU
INIT : Booting Build nr:48
WIFI : Connecting...
IP   : Static IP :192.168.5.101
INIT : I2C

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
ªU
INIT : Booting Build nr:48
WIFI : Connecting...
IP   : Static IP :192.168.5.101
INIT : I2C
</code>

I try different reset option...[ 10k to gpio0 gnd to reset, reset button different ways.. ]
I do have 3 esp boards and all 3 broken, when i did flash wit espeasy R48, ide 1.6.5

Is there some one that have working option for this problem?

marvinroger, 2016-01-25:
@Dylantje report your issue at esp8266/Arduino, this is not the place to do it as Homie has nothing to do with it. ;)

marvinroger, 2016-06-19:
The v2.0.0 will rely on the 2.3.0 version of the ESP8266 for Arduino. This problem is solved in this version.

closed 2016-06-19 by marvinroger

jmozmoz, 2018-05-13:
Eventhough, this bug has be closed sometime ago, here an observation, which might help others:
I had the following added to my platformio.ini and I observed the resets described above (about 5 s after startup). After I have removed the debug options, no more resets:

```
build_flags = -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_SSL -DDEBUG_ESP_TLS_MEM 
 -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_UPDATE
```