homieiot/homie-esp8266 #10: Android app to setup configuration

enavarro222, 2016-01-29:
Just an idea: It may be very convenient to have a lightweight android app that can setup Homie device config !

I'm not used to that, but I guest that https://github.com/marvinroger/homie-esp8266/tree/gh-pages can easily be pack in an adroid (or iphone) app ? With lib like cordova (https://cordova.apache.org/)

marvinroger, 2016-01-30:
Well in the future this may be a great addition. But for now you only have to open your browser and go to http://marvinroger.github.io/homie-esp8266/ - just fav it and you're ready to do it in seconds -.

There are other priorities for now. ;)

enavarro222, 2016-01-30:
For sure, this is not urgent.
I create the ticket to keep it in mind.

furyfire, 2016-02-10:
Here you go @enavarro222 
[Phone Gap build for anroid](https://build.phonegap.com/apps/1896165/share) 
See #21

marvinroger, 2016-02-10:
Oh, that's great! I'll need to add a `configurator` branch, so that a CI can build assets and publish it to `gh-pages`, before automatically building the apps on the PhoneGap build servers. 

But before I need to finish the configurator, as you can see it is not yet good looking on mobile.

furyfire, 2016-02-10:
I only tested it on my 7 inch nexus tablet. The only issue I stumbled across was the OTA not being configurable besides an on/off toggle

marvinroger, 2016-02-10:
Yes, it is in a usable form but is still a WIP. ;)

marvinroger, 2016-02-10:
It should be complete and responsive, now. Next step is to enable the CI.

marvinroger, 2016-02-10:
Alright: https://build.phonegap.com/apps/1906578/share

The configurator sources are on the `configurator` branch. When pushed to, the CI automatically builds and publishes the distribution version to the `gh-pages`, so it is accessible at http://marvinroger.github.io/homie-esp8266. Then, it tells PhoneGap Build to trigger a build of the Android and Windows Phone apps. So there is no overhead on our side maintaining the configurator: if you want to collaborate on it, just push on `configurator` and everything is handled under the hood. :)

enavarro222, 2016-02-10:
Whaou ! so nice ! I'll test that tonight.

marvinroger, 2016-02-10:
Please do! I think the circle is complete, now it's time to tell if the whole thing is stable... Or not.

marvinroger, 2016-02-10:
By the way, thanks @furyfire for the PhoneGap Build trick, I didn't know there was a cloud service available for that.

closed 2016-02-11 by marvinroger