homieiot/homie-esp8266 #11: MQTT not working if more than 4 subscriptions

enavarro222, 2016-01-30:
I got `Homie.sendProperty` that returns false (even if some time msg is sent) and some random MQTT reconnection when I get more than 4 `Homie.addSubscription`.

It seems to be linked to https://github.com/knolleary/pubsubclient/issues/98

Maybe one should add a `mqtt.loop` call in `Homie.addSubscription` ?

marvinroger, 2016-01-31:
Yes this is probably the issue. We need to add `mqtt.loop()` after https://github.com/marvinroger/homie-esp8266/blob/master/src/Homie/Boot/BootNormal.cpp#L99

I will fix tomorrow!

marvinroger, 2016-01-31:
Can you try again?

enavarro222, 2016-01-31:
Thanks, I'll test it tonight.

enavarro222, 2016-01-31:
OK you can close it, the fix works.

enavarro222, 2016-01-31:
Oh I didn't take care: I can close it myself...

commit 5c059c82 (2016-01-31): Attempt to fix #11
closed 2016-01-31 by enavarro222