homieiot/homie-esp8266 #15: NTP time, comments

furyfire, 2016-02-02:
I begun cleaning up my NTP code for Homie. This allows homie to receive the current time from an NTP server and thereby timestamp measurements or schedule future events.

I suggest adding another entry to the config setting the hostname of an NTP server and a new MQTT topic called $tzoffset which lets the server manage the timezone calculation. 
(I think this has to be server side to manage DST instead of just storing it as a fixed configuration)

Any other comments on adding time/date handling to Homie? 
Is this a bad idea?

marvinroger, 2016-02-02:
Thinking more about it, don't you think it's the server responsability to handle the timestamping measurements and scheduling of events? Clients have to be the most lighweight possible, both about computation and network bandwidth. How would you actually timestamp a message? By sending the timestamp in the message? It feels like a waste of bandwidth, when the server can do it. And more bandwidth means more transmitting time, means more power usage, means more cost. It is obviously negligible, but it deserves to be discussed.

furyfire, 2016-02-02:
Very valid. Not to mention this feature is fairly simple to implement on the sketch level.

marvinroger, 2016-02-07:
@furyfire may I close, no regrets? I want Homie for ESP8266 to stay as lightweight as possible, and the devices to be clients, not schedulers.

furyfire, 2016-02-07:
You may. The logic is the extremely easy to implement as a simple sketch also. There's is no need for this in homie core

marvinroger, 2016-02-07:
If at some point, more people ask for it, I'll reconsider. :) thanks!

closed 2016-02-07 by marvinroger