homieiot/homie-esp8266 #16: Move Config to filesystem

furyfire, 2016-02-03:
As the configuration array continues to grow and there is currently very little control validation on the data stored in EEPROM (checksums, lengths etc) would it not be an idea to simply store the json config in the file system?
https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md

It would simplify development as the flashtool today can automatically include the file. 

Validation of the config file could then be put in a shared module allowing Homie to drop back into config mode if the data is unusable. 

We would also retrain backwards compatibility to some extend as the JSON format would be easier to parse with the already included ArduinoJSON lib

Adding an optional setup call, which would default to
Homie.setConfig("home/config.json");

marvinroger, 2016-02-03:
I completely agree with this. And this would allow the developer to bypass the configuration mode, as this can be annoying if deploying a lot of devices. It will also simplify the API (`reserveEeprom` and `getEepromOffset` not needed anymore).

So:

- Check if `homie/config.json` exist
  - if it does, check it. If valid, boot into `normal` mode, else boot into `config` mode
  - if it does not, boot into `config` mode

Does it sound good?

enavarro222, 2016-02-03:
Sounds really good !

furyfire, 2016-02-03:
And if homie/ota exists boot into ota 
If the config file is invalid it should boot into config mode with the choice of downloading old config (so we have a future upgrade path)

marvinroger, 2016-02-03:
The choice of downloading old config? You can simply send the old json to the `/config` endpoint. I am not sure I understad what you mean!

furyfire, 2016-02-03:
During an upgrade to a newer release of Homie your old config file could become outdated. I was wondering if it would be useful for a GUI configuration tool to download the old config and be able to fix the issues behind the curtain before reapplying it to /config.

marvinroger, 2016-02-03:
The config file format won't change that often when we go stable, only on breaking releases. This is an interesting idea for later, though!

marvinroger, 2016-02-05:
It's stable on my side, could you please test?

marvinroger, 2016-02-06:
Feel free to reopen if something is wrong!

closed 2016-02-06 by marvinroger