homieiot/homie-esp8266 #17: Remove dependencies

marvinroger, 2016-02-06:
As suggested by @furyfire, it would be good to remove the dependencies from the library. It was impossible before because of PubSubClient that would not accept a method as its callback function, so I had to modify the library. It is now possible since knolleary/pubsubclient@98a9c296f6d469a9b6221eb2e29655610b4d7583 , and released in version 2.5.

So, to work, Homie would require PubSubClient >= 2.5, Bounce2 >= 2.0 and ArduinoJson >= 5.0.8 to be installed. The problem is there is no way to declare dependencies for Arduino libraries. Users would have to download 4 libraries, instead of just one.

Maybe just provide in the installation guide, the links to the 3 `.zip` of the 3rd party libraries? This is an overhead, but this is definitely cleaner than just embedding the libs.

What do you think? @enavarro222

furyfire, 2016-02-06:
I'm behind you on this. Mostly because I use platformio which makes it fairly easy to download required libs. I know the platformio project is brewing on building a framework for dependency management under embedded development (think npm, bower and composer for embedded)

marvinroger, 2016-02-06:
I know, I advised them to use semver. But until 3.0 is released, there is no reliable way to declare dependencies. And, PlatformIO is not very used yet.

enavarro222, 2016-02-06:
Nice ! Will be clearly better without copy of dependency.

With arduino IDE library manager it's really trivial to install dependency (I'm quite sure that the 3 needed are available with the default config).

So yes, one step more to have a basic Homie example working. But it's a small one.

Another point : I'got a Homie device that also use Bounce2. If one include it naively (from arduino lib. dir) it do not compile (conflict with the homie/3rd/bounce)... Same could appear with the JSON lib.

marvinroger, 2016-02-06:
Yes, I did not managed to import libraries into the `Homie` namespace, this is one of the reasons why I want not to embed dependencies.

marvinroger, 2016-02-06:
Done in 9644a8cc85ecd46cece591e92f5d5c90328d4ece. I added a note in the Wiki, about dependencies.

closed 2016-02-06 by marvinroger