homieiot/homie-esp8266 #18: Consider go stable

marvinroger, 2016-02-06:
The API is quite complete and the overall stability is good (didn't run into any troubles so far, after weeks of continuous use). Docs are up. I guess it's almost time to release a stable version.

But before:

- [ ] We need to create a Web UI to configure the device, compatible with the latest API

furyfire, 2016-02-06:
During config mode there should be a way to fetch homie version, firmware name and firmware version.
The web interface does not currently have any means of reading out this data.

marvinroger, 2016-02-06:
Good point. What about a GET `/info` endpoint? Maybe `/device-info`? The latter sounds cooler.

marvinroger, 2016-02-07:
The nodes might also be exposed, like so:

```json
{
  "homie_version": "1.0.0",
  "firmware": {
    "name": "awesome-light",
    "version": "1.0.0"
  },
  "nodes": [
    {
      "id": "light",
      "type": "light"
    }
  ]
}
```

marvinroger, 2016-02-07:
@furyfire , latest git rev exposes a GET `/device-info` that responds the above json plus a `device_id` property. Anything else?

furyfire, 2016-02-07:
Amazing work!
 I'll be back at my desk with a nodemcu kit tomorrow afternoon to test it all!

marvinroger, 2016-02-29:
@furyfire @enavarro222 Guys, I've been using it in my home for weeks without interruption, and so far everything is good. I am definitely going to bump it to 1.0.0, but if you are using it, did you encounter any stability issues?

furyfire, 2016-02-29:
All good for me. The only issues I have are with SSL since I still use arduino esp8266 2.0

Anyone else using platformio?

marvinroger, 2016-02-29:
You can use 2.1.0, it is released on the stable channel now, as you wanted!

I am not at this time, but the library is available on pio.

marvinroger, 2016-02-29:
And 2.1.0 [is available on pio](https://twitter.com/platformio_org/status/704304862998224896)

enavarro222, 2016-02-29:
I got a working firmware for a NeoPixel based esp8266 lamp for two weeks... But it's still on my shelve, I got no time to finish it (hardware staff) and install it "in prod". So no real test for now.

However everything seems nice. I use events with a custom Homie status notification, just work's fine. I just notice a limitation with MQTT msg size, but that's an issue with PubSubClient not Homie. No time to investigate it further for now...

furyfire, 2016-03-01:
Can confirm SSL is working as of esp8266-Arduino 2.1.0 :+1:

marvinroger, 2016-03-04:
Great! v1.0.0 on its way!

closed 2016-03-04 by marvinroger