homieiot/homie-esp8266 #23: Serial conflict with BUILTIN_LED for generic ESP8866

enavarro222, 2016-02-10:
I got a tricky serial issue with a generic ESP module... I eventually figure that for "generic ESP8866 module" the BUILTIN_LED is set to 1 in arduino/esp8266 (https://github.com/esp8266/Arduino/blob/573a0fb47fc0dc6dc25010d50642494dbb069476/variants/generic/pins_arduino.h#L45)
and GPIO 1 is nothing else that serial TX... So on startup serial was not working well.

Not sure how to fix that. We may add a way to configure the led to use ?
Also I'm thinking on how to make "state indicator" much more flexible... I got a lamp with some "NeoPixel" and before Homie I just made one of the pixel to blink to indicate pending wifi connection and so on. (Btw such flexibility can wait a future release !)

marvinroger, 2016-02-10:
Oh, that's a problem... So yes, this should be configurable. What about an event system?

```c++
#include <Homie.h>

void onHomieEvent(HomieEvent event) {
  switch(event) {
    case HOMIE_CONFIGURATION_MODE:
      // Do whatever you want when configuration mode is started
      break;
    case HOMIE_NORMAL_MODE:
       // Do whatever you want when normal mode is started
      break;
    case HOMIE_OTA_MODE:
       // Do whatever you want when OTA mode is started
      break;
    case HOMIE_ABOUT_TO_RESET:
      // Do whatever you want when the device is about to reset
      break;
    case HOMIE_WIFI_CONNECTED:
       // Do whatever you want when Wi-Fi is connected in normal mode
      break;
    case HOMIE_WIFI_DISCONNECTED:
      // Do whatever you want when Wi-Fi is disconnected in normal mode
      break;
    case HOMIE_MQTT_CONNECTED:
      // Do whatever you want when MQTT is connected in normal mode
      break;
    case HOMIE_MQTT_DISCONNECTED:
      // Do whatever you want when MQTT is disconnected in normal mode
      break;
  }
}

void setup() {
  Homie.enableBuiltinLedIndicator(false);
  Homie.onEvent(onHomieEvent);
}
```

This would remove the need for `Homie.setResetHook()`, as this would allow to hook to every valuable Homie events.

I like the idea of an event system because of what it allows. Thinking about the `NeoPixel`, I told myself that you would also like to change the color of a LED on the strip depending on the mode the device is on.

enavarro222, 2016-02-10:
Event system could be nice !
However not straightforward if you want to make a led/pixel to blink in some cases. Maybe with ticker and some global state variables...

enavarro222, 2016-02-10:
Btw for `NeoPixel` devices I'm thinking on MQTT convention to use, I'll start an issue for that.

marvinroger, 2016-02-11:
Not straightforward?

```c++
#include <Homie.h>
#include <Ticker.h>

Ticker led_ticker;

void blinkTick() {
  digitalWrite(CUSTOM_LED, !digitalRead(CUSTOM_LED));
}

void blink(bool start) {
  if (start) {
    digitalWrite(CUSTOM_LED, HIGH);
    led_ticker.attach(1000, blinkTick);
  } else {
    led_ticker.detach();
    digitalWrite(CUSTOM_LED, LOW);
  }
}

void onHomieEvent(HomieEvent event) {
  switch(event) {
    case HOMIE_WIFI_CONNECTED:
      blink(true);
      break;
    case HOMIE_WIFI_DISCONNECTED:
      blink(false);
      break;
  }
}

void setup() {
  Homie.enableBuiltinLedIndicator(false);
  Homie.onEvent(onHomieEvent);
}
```

The problem is if I want to make it more straightfoward, I would have to provide `setLedPin`, `setLedState`, `setWifiLedPace`... that's too much noise.

enavarro222, 2016-02-11:
Sounds great (and yes straightforward) !
I never used ticker for now.

closed 2016-02-11 by marvinroger

enavarro222, 2016-02-11:
great ! thanks !

marvinroger, 2016-02-11:
The advanced usage page in the Wiki is updated, also. ;)

commit d5a8a5e2 (2016-02-11): Implement event system, fix #23