homieiot/homie-esp8266 #24: MQTT convention for `NeoPixel` lamp/device

enavarro222, 2016-02-10:
I'm thinking how to control a device made of an ESP8266 that drive some `NeoPixel` (or WS2801).

For now I got:
* `devices/ID/state/set` with `on`, `off` or `toggle`
* `devices/ID/rgb/set` with color eihter in `RRGGBB` html-like hex format or `rrr,ggg,bbb` decimal format. This control the color of *all* leds.

At first I only use hex format (much more easy to parse) but then I implemented decimal&coma one to be compatible with Home Assistant (https://home-assistant.io/components/light.mqtt/).

The missing point here is a way to control each led individually...

Two idea:
* `devices/ID/rgb_<led_id>/set` with color of the given led ID
* `devices/ID/rgb_each/set` with one msg giving the color for each led. In a first implementation I used just the concatenation of `RRGGBB` html-like hex format for each led.

What do you think about all that ?

A other way is to use json in MQTT msg... but I'm not sure about performances.

It could be nice to have a script that push MQTT msg to get effects as here : https://www.youtube.com/watch?v=OLxM9p5o2us

marvinroger, 2016-02-11:
If you have something like 3 RGB leds, the first solution would be the best fit. Now, if you want to set the color of a whole strip, the second solution would be better (devices/ID/strip/state/set):

* You only have to declare one node
* There is only one subscription done on the MQTT broker, so it is faster
* When you receive a payload, you receive it in one packet with the MQTT overhead. It is way faster and lighter

So the concatenation is the best idea to get the result of the YouTube video.

But there is one thing you need to know: I don't want to put a convention on Homie types of nodes. Because everybody will have its own requirements. (as you said, you are using Home Assistant, maybe this won't work for people using OpenHab).

marvinroger, 2016-03-04:
Feel free to reopen if the above answer was not sufficient. ;)

closed 2016-03-04 by marvinroger