homieiot/homie-esp8266 #25: Configurator, better use IP than `homie.local` ?

enavarro222, 2016-02-11:
I just test the configuration app (both as a webpage/or "native" app) with a pretty old android phone. It didn't work :( I guess there is a issue "homie.local" with mdns resolution...

Anyway, in the configuration tool it may be preferable to use the default ip ?

I have to fix my phone screen before to test on a more recent android device :/

marvinroger, 2016-02-11:
The configuration app makes use of the `fetch` API, which is only available in Android WebView >= 42. Don't you have a Play Store update for the Android System WebView or something?

Anyway, there is no mdns resolution, it is normal dns (dns server built in the `configuration` mode), so this is not the issue. A solution to this would be to use a more recent web runtime, like [Crosswalk](https://crosswalk-project.org/), but the app will take much more space.

enavarro222, 2016-02-11:
Ok thanks !
So update in progress...

marvinroger, 2016-02-11:
When you say update in progress, do you mean the Android System WebView update? Does it work?

If I embed the Web runtime, the apk will be 20MB instead of only 1MB... But if it's needed, why not, after all.

enavarro222, 2016-02-11:
Here is a more informative report. I'm not familiar with android "internals".

I test with two phones.

One run android 4.1.1. I can't find any "Android System WebView" in application list. Only a "visionneuse HTML" (in french) in version 4.1.1. Here the app doesn't work. No error msg. It just get stuck in "Waiting for the device..." screen. Same when tested from android navigator. However it works with Chrome v48 (didn't test yet with FF).

Then a second phone android 5.1.1 (Cyanogen) with  "Android System WebView" version 39. Same issue than above. Here I test with FF and it works (v 41.0.2).

So yes clearly it needs a recent Android WebView (as you say >= 42). At least it could be nice to have a version checking during install...

marvinroger, 2016-02-11:
Could you please test with the [latest APK](https://build.phonegap.com/apps/1906578/share)? It's 20 MB  but I can't do otherwise.

enavarro222, 2016-02-11:
It works ! nice :)
(and the page layout is ok, it was not in the 1st version)

So now the only problem may be with some browser...

marvinroger, 2016-02-11:
About the layout, this is because of flex.

Most people are using modern browsers, so it should not be a problem. Especially devs!

enavarro222, 2016-02-11:
For sure.
So I close the issue for now. Thanks again !

closed 2016-02-11 by enavarro222