homieiot/homie-esp8266 #28: Any way to auto discover server?

mishoboss, 2016-03-14:
Hi, first - really great project! Huge congrats!
Question - is there a chance for SSDP or other way for nodes to auto discover server IP address in the local network?

marvinroger, 2016-03-14:
Thanks a lot! 

I would go for mDNS as the built-in library for mDNS supports querying whereas the SSDP one does not. What about, in the [JSON configuration file](https://github.com/marvinroger/homie-esp8266/wiki/5.-JSON-configuration-file), a new `mdns` field in both the `mqtt` and `ota` that, if set to `true`, would query for the mDNS service defined in a new `service` field (making `host` and `port` optional). I hope I was clear!

Does it sound good to you?

mishoboss, 2016-03-14:
It sounds even more than good! :)

marvinroger, 2016-03-14:
Great! I'll implement it as soon as I can. ;)

lvjh, 2016-03-15:
Arduino:1.6.8 (Linux), 开发板："Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

/home/esp/Arduino/libraries/homie-esp8266-1.1.0/src/Homie/Boot/BootOta.cpp: In member function 'virtual void HomieInternals::BootOta::setup()':
/home/esp/Arduino/libraries/homie-esp8266-1.1.0/src/Homie/Boot/BootOta.cpp:46:188: error: no matching function for call to 'ESP8266HTTPUpdate::update(char [64], uint16_t&, char [64], String&, bool&, char [60], bool)'
   t_httpUpdate_return ret = ESPhttpUpdate.update(Config.get().ota.host, Config.get().ota.port, Config.get().ota.path, dataToPass, Config.get().ota.ssl, Config.get().ota.fingerprint, false);
                                                                                                                                                                                            ^
/home/esp/Arduino/libraries/homie-esp8266-1.1.0/src/Homie/Boot/BootOta.cpp:46:188: note: candidates are:
In file included from /home/esp/Arduino/libraries/homie-esp8266-1.1.0/src/Homie/Boot/BootOta.hpp:5:0,
                 from /home/esp/Arduino/libraries/homie-esp8266-1.1.0/src/Homie/Boot/BootOta.cpp:1:
/opt/arduino-1.6.8/hardware/esp8266com/esp8266/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h:52:29: note: t_httpUpdate_return ESP8266HTTPUpdate::update(const char*, const char*, const char*)
         t_httpUpdate_return update(const char * url, const char * current_version = "", const char * httpsFingerprint = "");

                             ^
/opt/arduino-1.6.8/hardware/esp8266com/esp8266/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h:52:29: note:   candidate expects 3 arguments, 7 provided
/opt/arduino-1.6.8/hardware/esp8266com/esp8266/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h:53:29: note: t_httpUpdate_return ESP8266HTTPUpdate::update(const char*, uint16_t, const char*, const char*, bool, const char*)
         t_httpUpdate_return update(const char * host, uint16_t port, const char * url = "/", const char * current_version = "", bool https = false, const char * httpsFingerprint = "");

                             ^
/opt/arduino-1.6.8/hardware/esp8266com/esp8266/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h:53:29: note:   candidate expects 6 arguments, 7 provided
In file included from /home/esp/Arduino/libraries/homie-esp8266-1.1.0/src/Homie/Boot/BootOta.hpp:5:0,
                 from /home/esp/Arduino/libraries/homie-esp8266-1.1.0/src/Homie/Boot/BootOta.cpp:1:
/opt/arduino-1.6.8/hardware/esp8266com/esp8266/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h:54:29: note: t_httpUpdate_return ESP8266HTTPUpdate::update(String, uint16_t, String, String, bool, String)
         t_httpUpdate_return update(String host, uint16_t port, String url = "/", String current_version = "", bool https = false, String httpsFingerprint = "");

                             ^
/opt/arduino-1.6.8/hardware/esp8266com/esp8266/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h:54:29: note:   candidate expects 6 arguments, 7 provided
exit status 1
Error compiling for board Generic ESP8266 Module

lvjh, 2016-03-15:
I WANT USE HOMEI ，BUT COMPILE ERROR ,CAN YOU HELP ME?

marvinroger, 2016-03-15:
I bet you are using the 2.0.0 version of the Arduino for ESP8266, you need >= 2.1.0.

By the way, please fill a new issue if you have a problem, instead of pulluting this one. Thanks!

marvinroger, 2016-03-20:
Unfortunately, this will have to wait the next Arduino for ESP8266 as mDNS query support has been added after the 2.1.0 release.

mishoboss, 2016-03-20:
BTW, as far as I know there ahould be installed mDNS/bonjour servers on the Node.js side in order to work. I mean, the solution is not purely node, but we rely on external daemon/service for different OS. Is this the case?

marvinroger, 2016-03-20:
It seems like there is a pure JS implementation : https://github.com/mdns-js/node-mdns-js

marvinroger, 2016-04-19:
Will land in next version!

closed 2016-04-19 by marvinroger

marvinroger, 2016-04-23:
Released!