homieiot/homie-esp8266 #29: Static IP support

marvinroger, 2016-03-17:


marvinroger, 2016-03-20:
Out of scope, DHCP is a better fit, and IP can be configured per device on router interface. Feel free to reopen with your use case if needed.

CWempe, 2016-03-20:
I prefer DHCP, too.
Unfortunately I cannot change the IP of my devices on my dhcp server.

One of the few shortcomings of the biggest (and best) DSL-Router brands in germany (www.avm.de) does not allow you to change the IP address of dhcp clients on its web interface (or elsewhere). :(

While everything works well now, it would still be neat to change the IP on the device.
Maybe reopen this feature request wioth low prioroty? :)
Unless I am the only one how would benefit from this feature.

marvinroger, 2016-03-20:
Alright, then. :)

claybar, 2016-09-02:
@CWempe:
> One of the few shortcomings of the biggest (and best) DSL-Router brands in germany (www.avm.de) does not allow you to change the IP address of dhcp clients on its web interface (or elsewhere). :(

I think there is a way, although not the nicest.  Once a device is connected to the router, it can be selected and the assigned IP can be reserved for future.  This means subsequent connections will receive the same IP.  Also, before a device is connected a IP can be reserved by pre-entering the MAC address and choosing an IP.

CWempe, 2016-09-02:
@claybar 
> Also, before a device is connected a IP can be reserved by pre-entering the MAC address and choosing an IP.

I did not know this feature existed!
Maybe it is new or I never noticed it. :(

Nevertheless, that solves my problem. :+1: 
I just need to delete the "wrong" IP and enter the new entry.

Thanks!

marvinroger, 2016-09-02:
This was kind of surprising that you did not have this feature, well glad you found out how to use it.

marvinroger, 2016-09-03:
Reclosing it, then!

closed 2016-09-03 by marvinroger

euphi, 2016-09-05:
There are many routers where you can't set the IP for DHCP clients.

However, I still don't see a use case for a static IP for Nodes as long as MQTT is the only communication protocol.

marvinroger, 2016-09-06:
I agree, it could have been useful for some OTA implementations on the ESP8266, but as OTA is handled over MQTT, I don't see a use case either.

shogsbro, 2016-09-06:
I wonder how you guys do your firmware development. 

I initially used Arduino with OTA, and progressed to platformio once I realised how much better it is. Both support OTA programming, and since I use OpenWrt on my router I don't have any problem doing a static IP reservation so I can easily specify the upload target.

Do you use a shim to send the uploading firmware via @jpmens python framework for uploading?

marvinroger, 2016-09-06:
I update via USB (512 000 bauds), it only takes like 3-4 seconds. And if I need to update over OTA, I do it over MQTT, as it is fastest in my opinion.

shogsbro, 2016-09-06:
OK, for MQTT upload do you use a custom uploader command-line, using mosquitto_pub? It would be helpful to include documentation on this for noobs.

marvinroger, 2016-09-07:
I don't this too often so I do use `mosquitto_pub`. I'll enhance the docs when everything else is done.

johnjore, 2016-09-10:
I've just started tinkering with ESP8266's and came across this framework. Looks very good and timesaving, but I use static IP addresses for these items. I actually have DHCP setup in a HA configuration using VMs, but I want my sensors to be independant of the IT infrastructure. My other half would be less than impressed if I setup the ESP to control things like sockets and switches and the ESXi server, or its iSCSI storage, dies and the sockets/switches stopped working... (This has already happend, the storage controller died and it took 4 weeks to get it replaced under warranty. I now have a spare on a shelf...)

Yes, I could setup another DHCP on RaspPi or a spare WLAN box, but why add more infrastructure / components when I can just configure each device with a static IP and not rely on DHCP or DNS?

I've added 
```
void onHomieEvent(HomieEvent event) {
  switch(event) {
    case HomieEvent::WIFI_CONNECTED:
      WiFi.config(ip, gw, netmask);
      Serial.print("IP Address: ");
      Serial.println(WiFi.localIP());
    break;
```
Having the IP as a configuration setting would be better and maybe one day I modify this to read the IP from a file on the SPIFFS, but best if Homie framework itself supported the use of static IPs.

Just my 2c and please keep up the good work. 

JJ

monster1025, 2017-03-31:
It will be great to have option to specify static IP - because in low-power mode (when powered on battery) - wake up with DHCP takes ~2sec, but with static ip is only 200ms (and it saves battery power).

marvinroger, 2017-04-19:
Implemented in https://github.com/marvinroger/homie-esp8266/pull/327 👍