homieiot/homie-esp8266 #31: support wired networks

CWempe, 2016-03-17:
It would be great if Homie would support wired networks, too.

Like the ENC28J60 chip.
https://www.microchip.com/wwwproducts/en/en022889

This might be usefull for cases where the wifi signal is to weak.

marvinroger, 2016-03-17:
Thinking about it, how would we implement the config mode? Would it even exist?

CWempe, 2016-03-17:
There are my thought on this topic.
Just off the top of my head.

Since you usually don't need authentication for wired networks the device should connect to the existing network without problems and get an IP via DHCP.
The (Web)App could discover the device via mDNS or the user could enter the IP.

The user could configure the parameters via Serial.

Optionaly hardcode the parameters in the sketch.

The device could have an initiating ip address (like 10.1.1.1) and the user could connect by configure a corosponding ip on his computer. (not the easiest way)

marvinroger, 2016-03-17:
> The (Web)App could discover the device via mDNS or the user could enter the IP.

For the mDNS part, no it can't, a web app can only communicate through HTTP. This would require native code. But implementing mDNS would be an interesting idea for 3rd party configurators using native code. The user could also enter the IP, that's right.

> Optionaly hardcode the parameters in the sketch.

What is good about Homie sketches is that they are portable. There are no hardcoded credentials, so you can share the firmwares. I don't want to break this flexibility.

> The user could configure the parameters via Serial.

This is already possible, you can upload `/homie/config.json` to the SPIFFS.

I am not sure when I will implement this, if I ever. This is a big refactoring and even the name would need to change (Homie for ESP8266 -> Homie for Arduino).

CWempe, 2016-03-17:
To be clear: I am not planing to use wired devices.

It's just a (valid) idea that came up, when I discussed it with a friend who has many ethernet cables in his house. ;)

marvinroger, 2016-04-02:
I tried to picture how I could implement this, and the conclusion I came to was that it is not feasible in the current form of the project. On wired devices (in other words, genuine Arduino), there is no SPIFFS interface, no SSL support, and most importantly not enough RAM. The project would need to be rebuilt from the ground up, and I don't have enough resources to. This might be a good idea for a fork, though!

marvinroger, 2016-06-02:
Definitely out of the scope of the project. Once again, a fork implementing this would be great!

closed 2016-06-02 by marvinroger