homieiot/homie-esp8266 #33: reset remotely via mqtt

CWempe, 2016-03-17:
How about a specific mqtt topic to reset the device remotely?

like:
devices/xXxXxXxX/$reset true
or
devices/15b638e0/$online/set true

marvinroger, 2016-03-17:
Already done, see https://github.com/marvinroger/homie/blob/master/README.md#device-properties

CWempe, 2016-03-17:
I admit, I did not see that. :)

But this does reset the configuration.
What I meant was just a reload/restart/reboot of the device.

marvinroger, 2016-03-18:
If this is useful in the case of your sensor, you can implement it yourself easily (just create a node that you subscribe to a `restart` property). As you said, this would be useful to fix the openHAB behavior, and I don't see any other use cases to that. So once again, I am not in favor for implementing it. I hope you understand my point of view ;)

CWempe, 2016-03-18:
I do understand.
Feel free to close these issues. :)

closed 2016-03-19 by marvinroger