homieiot/homie-esp8266 #34: force resending states

CWempe, 2016-03-17:
When you change your configuration or restart http://www.openhab.org/ the current states of your device might get lost (Yes, you can prevent this).

But it would be nice to send a mqtt message to let the device resend the initial messages.

devices/15b638e0/$*

An alternative workaround would be to reset the device remotely (https://github.com/marvinroger/homie-esp8266/issues/33)

CWempe, 2016-03-17:
I think a feature to resending specific values like temperature (rather than wait for the next TEMPERATURE_INTERVAL) need to be implemented by the user.

marvinroger, 2016-03-17:
This is already handled by MQTT, and the initial messages are sent with the retain flag set to true, which means that when a new client connect it receives all the latest values sent by the device.

CWempe, 2016-03-17:
This would not help in my case, since openhab can lose the values while still keep up the connection to the mqtt broaker.

But like I said, this is not an issue when openhab ist configured correctly.
And I dont want you to break the mqtt standards. :)

marvinroger, 2016-03-18:
I am not in favor of that, I don't want to workaround a bad behavior of a 3rd party software. openHAB should fix the issue on its side, don't you think? And as you said, this is not an issue if it's configured correctly. ;)

closed 2016-03-19 by marvinroger