homieiot/homie-esp8266 #35: [Wiki] How to setup the OTA web server?

CWempe, 2016-03-17:
I still don't understand how to setup the ota web server.

I understand how it technically works.
But do I have to develope this myself?

Or is there some kind of index.php file that I can copy to my apache web server in a "ota" subfolder?

marvinroger, 2016-03-18:
This is very DIY, I admit! You have an example PHP script on the [Arduino for ESP8266 Wiki](http://esp8266.github.io/Arduino/versions/2.1.0/doc/ota_updates/ota_updates.html#http-server). Or if JS is more your thing, here is an OTA script I wrote: https://github.com/marvinroger/homie-server/blob/master/lib/servers/ota.js#L126

marvinroger, 2016-03-18:
I added the above links on the Wiki.

marvinroger, 2016-03-19:
Links also added on the Wiki.

closed 2016-03-19 by marvinroger

rschaten, 2016-03-28:
I just built myself a simple OTA server, based on the PHP-example in the wiki. It works, but I thought that the version-header (`x-ESP8266-version`) would include the version the node would like to be upgraded to (the one I wrote to `$ota` via MQTT). There's always the currently running version in there.

Is it possible to ask the OTA server for a certain version of the firmware? Meaning: is it possible to let the server know which version was announced via MQTT?

marvinroger, 2016-03-28:
I will implement this @rschaten ;)

rschaten, 2016-03-28:
Thanks! :+1: