homieiot/homie-esp8266 #40: When may setNodeProperty() be invoked?

jpmens, 2016-03-20:
First of all, thank you for building Homie-ESP8266; this is grand.

Assuming I want to add a property to a node to indicate the unit, I tried doing

```
Homie.setNodeProperty(tempnode, "unit", "C", true);
```

in `setup()`, just after `registerNode()`, just after, and after `Homie.setup()`, none of which works. The only place I seem to succeed in seeing this property is if I add it to a loop handler, which is a shame,  as it's a one-time only constant property.

Am I holding it incorrectly? :-)

marvinroger, 2016-03-20:
Thanks a lot. :)

Perfect timing, I've just updated the [temperature example](https://github.com/marvinroger/homie-esp8266/blob/master/examples/TemperatureSensor/TemperatureSensor.ino) with what you want! What you need is `setSetupFunction()`.

To answer to your question, the `setNodeProperty()` function can be invoked when the device is in normal mode and connected to both the Wi-Fi and MQTT broker. The `setupFunction` and `loopFunction` that you can register are called when this condition is true.

jpmens, 2016-03-20:
I've built and uploaded your temperature example, but I don't get a `unit`:

```
devices/0d87bde0/$nodes temperature:temperature
devices/0d87bde0/$online true
devices/0d87bde0/$name Bookshelf
devices/0d87bde0/$localip 192.168.1.192
devices/0d87bde0/$fwname awesome-temperature
devices/0d87bde0/$fwversion 1.0.0
devices/0d87bde0/$signal 92
devices/0d87bde0/temperature/temperature 22.00
```

marvinroger, 2016-03-20:
Oh, you're right. Let me take a look at it.

jpmens, 2016-03-20:
I think it's the same issue I had earlier: a `Serial.print()` in setupHandler indicates it's being called *before* the MQTT connection is up.

marvinroger, 2016-03-20:
Could you please try again with the latest git rev?

jpmens, 2016-03-20:
Works like a charm:

```
devices/0d87bde0/temperature/unit c
devices/0d87bde0/temperature/aaaaaaa AAAA
```

Thank you for fixing this so quickly.

closed 2016-03-20 by jpmens

marvinroger, 2016-03-20:
Great, my pleasure!

commit 5957a65c (2016-03-20): Fix #40