homieiot/homie-esp8266 #41: LWM2M in homie-esp8266

jpmens, 2016-03-20:
I love MQTT so Home has about everything I need. Be that as it may, I recently learned of a CoAP-based standard called LWM2M ([overview slides](https://www.slideshare.net/mobile/zdshelby/oma-lightweightm2-mtutorial)).

This screenshot shows Leshaan in operation:

[image]

Do you think this would be a good fit for Homie-esp8266 in addition to MQTT? Using, say, [Leshaan](http://www.eclipse.org/leshan/) to control, and [Wakaama](https://github.com/eclipse/wakaama) as the client code, Homie-ESP8266-controlled devices could be managed with the OMA standard.

marvinroger, 2016-03-20:
I am not sure it would be a good fit for Homie for ESP8266 in its current form, as it targets exclusively MQTT. Handling both MQTT and LWM2M would be too heavy. But this could be a good case for a fork.

marvinroger, 2016-03-28:
I guess, after the stability issues we've seen with TLS, that we can agree it would be too heavy for our beloved ESP8266 chips. May I close this? :)

closed 2016-03-28 by jpmens