homieiot/homie-esp8266 #44: Make _mqttBaseTopic configurable?

jpmens, 2016-03-21:
Homie currently hard-codes the MQTT base topic to `"devices"`. I'd like to propose making this configurable via JSON:

```json
"mqtt" : {
  "port" : 1883,
  "basetopic" : "sensors/homie/",
  ...
}
```

`basetopic` would default to what it currently is (`"devices"`). Making it configurable would enable people to better integrate Homie devices into their MQTT topic tree.

I would give this a stab myself and provide a PR if you think it a good idea.

marvinroger, 2016-03-21:
I agree, go ahead if you want to, but I can do it if you want. :wink:

jpmens, 2016-03-21:
@marvinroger considering I'd have to ask you for C++ help (not C) in getting at the Configuration values, you'd better do it properly. :smile: I'll do the testing.

marvinroger, 2016-03-21:
I'll do it then. ;)

jpmens, 2016-03-21:
@marvinroger this was fast. Maybe too fast: it's stacktracing the device. (I could have probably done that too :laughing: )

marvinroger, 2016-03-21:
@jpmens my bad, I fixed the bug. :)

jpmens, 2016-03-21:
Almost. :-)

```
sensors/homie/cf3a07e0/$node�7�?�7�?42!@�4�?�4�?�4�?�C�? light:light,temperature:temperature,relay:relay
```

jpmens, 2016-03-21:
Device is still periodically "rebooting"; console shows MQTT reconnecting

jpmens, 2016-03-21:
There's also one "devices/" in there:

```
sensors/homie/cf3a07e0/$signal ignal
sensors/homie/cf3a07e0/$online false
sensors/homie/cf3a07e0/$node�7�?�7�?42!@�4�?�4�?�4�?�C�? light:light,temperature:temperature,relay:relay
sensors/homie/cf3a07e0/$online true
sensors/homie/cf3a07e0/$name Dtest
sensors/homie/cf3a07e0/$localip 192.168.1.193
sensors/homie/cf3a07e0/$fwname h-sensor
sensors/homie/cf3a07e0/$fwversion 1.0.1
devices/cf3a07e0/temperature/degrees 22.00
```

jpmens, 2016-03-21:
and `$signal` is broken. I'll shut up now and go away.

marvinroger, 2016-03-21:
Please don't, you're helping a lot! The fact is i've optimized the strings for `devices/`, so there is a lot of micro optimizations I forgot to update, which cause problems. And I am getting tired :laughing:

jpmens, 2016-03-21:
I can imagine you're tired; so am I, hence my saying I'd go away -- I was hoping to be able to relax a bit. I've pulled again: better, but not good:

```
sensors/homie/cf3a07e0/$nodes light:light,temperature:temperature,relay:relay
sensors/homie/cf3a07e0/$online true
sensors/homie/cf3a07e0/$name Dtest
sensors/homie/cf3a07e0/$localip 192.168.1.193
sensors/homie/cf3a07e0/$fwname h-sensor
sensors/homie/cf3a07e0/$fwversion 1.0.1
sensors/homie/cf3a07e0/$online false

```

Device is crashing somewhere perdiodically, but not consistently. After typing this, I see

```
sensors/homie/cf3a07e0/temperature/degrees 22.00
```

No `$signal`, so I'm guessing the crash is there somewhere? Console shows "sending initial informations", but doesn't mention signal.

marvinroger, 2016-03-21:
I am sorry, I thought you meant you'd go away and stop posting continuously! Let's continue this discussion later then.

I tried with the same base topic as yours, and I correctly receive $signal, and no "crash" either. I'll see more in depth tomorrow!

jpmens, 2016-03-22:
A question related to the crashes/reboots: should Homie/ESP8266-Arduino actually always restart the device? I.e. is there a watchdog which handles that? I'm asking b/c this morning I see the device is no longer booting/reporting so I assume it's hung itself up.

marvinroger, 2016-03-22:
Yes, there is a watchdog built into the ESP8266. But I never had a wdt (watchdog timer) issue with Homie. However, the MQTT connection is sometimes lost for a second or two, and then is up again. Wi-Fi interference, I guess.

jpmens, 2016-03-22:
Regarding the reboots, I'm seeing this on the console:

```
✔ Wi-Fi connected
⌔ Attempting to connect to MQTT
Sending initial informations
✔ MQTT connected
Calling setup function
Sending Wi-Fi signal quality
✖ MQTT disconnected
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
✔ MQTT connected
✖ MQTT disconnected
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
⌔ Attempting to connect to MQTT
Sending initial informations
...
```

marvinroger, 2016-03-22:
This is not a reboot, these are only reconnections. What's weird is that the `MQTT disconnected` message is not present everytime. Maybe a problem in the loop.

marvinroger, 2016-03-22:
Are you using SSL? If yes, are there reconnections without SSL?

jpmens, 2016-03-22:
TLS (SSL) was enabled. I've reconfigured to plain, and I'm not seeing reconnects (in the first minute or two). FWIW, `fingerprint = ""`.

jpmens, 2016-03-22:
I should add that TLS was very stable throughout yesterday, before the `base_topic` changes.

marvinroger, 2016-03-22:
So I think the issue comes from the `String` class, which consumes too much RAM, or fragment the memory. I will try to use a fixed buffer to fix this. Like yesterday, I am sorry I am unable to test the code on real hardware for now. :/

jpmens, 2016-03-22:
No problem at all; I am happy to contribute by testing!

marvinroger, 2016-03-22:
I removed almost every `String` and done a lot of optimizations, which I am pretty sure will break something, so I'll wait tonight to push the code. ;)

jpmens, 2016-03-22:
Go ahead and push now if you like -- I'm ready to test the catastrophe. ;-)

marvinroger, 2016-03-22:
It may actually break eveything, even the config mode... Alright, then!

jpmens, 2016-03-22:
I note a slight formatting glitch upon startup (port missing) -- this may not be new though: 

```
  • OTA
    • Enabled: yes
    • Host: 192.168.1.130
    • Port:     • Path: /ota
    • SSL: no
```

In TLS/SSL mode not much has changed: I still see mutliple reconnects.

In plain mode, stability has deteriorated. After about 1m, the device dies: broker logs: 

```
1458652350: Client Homie-cf3a07e0 has exceeded timeout, disconnecting.
1458652350: Socket error on client Homie-cf3a07e0, disconnecting.
```

Stragely: I then press reset, and I see a normal-lookup startup procedure in Arduino console, but there's no connection to the MQTT broker ... If I kill (stop) the broker and reset device, it says "MQTT connected" (which is not possible).

marvinroger, 2016-03-22:
As I said, I will have to debug the thing heavily! I will tonight.

jpmens, 2016-03-22:
Please forgive me: I was looking at the wrong broker when in plain mode. (_sighs_). I have edited my previous comment for it to make sense.

marvinroger, 2016-03-22:
Could you please try with the latest git rev?

jpmens, 2016-03-22:
No visible difference in plain mode in the first 2 minutes, which is good.

Now running in TLS mode:

```
 • OTA
    ��� Enabled: yes
    • Host: 192.168.1.130
    • Port: 80
    • Path: /ota
    • SSL: no
⌔ Attempting to connect to Wi-Fi
✔ Wi-Fi connected
⌔ Attempting to connect to MQTT
Sending initial informations
✔ MQTT connected
Calling setup function
Sending Wi-Fi signal quality
Temperature: 23.75 °C
✖ MQTT disconnected
⌔ Attempting to connect to MQTT
Sending initial informations
✔ MQTT connected
Temperature: 23.75 °C
Temperature: 23.75 °C
Temperature: 23.69 °C
...
```

1. A bit of "garbage" in the OTA configuration display which may be a problem
2. One disconnect from MQTT followed by a reconnect

Broker-side, I see the following sequence of events (`fingerprint = ""`):

```
1458679257: New client connected from 192.168.1.192 as Homie-cf3a07e0 (c1, k15).
1458679257: Sending CONNACK to Homie-cf3a07e0 (0, 0)
1458679257: Received PUBLISH from Homie-cf3a07e0 (d0, q0, r1, m0, 'sensors/homie/cf3a07e0/$nodes', ... (46 bytes))
...
1458679257: Received SUBSCRIBE from Homie-cf3a07e0
1458679257: 	sensors/homie/cf3a07e0/$reset (QoS 1)
1458679257: Homie-cf3a07e0 1 sensors/homie/cf3a07e0/$reset
1458679257: Sending SUBACK to Homie-cf3a07e0
1458679257: Received SUBSCRIBE from Homie-cf3a07e0
1458679257: 	sensors/homie/cf3a07e0/$ota (QoS 1)
1458679257: Homie-cf3a07e0 1 sensors/homie/cf3a07e0/$ota
1458679257: Sending SUBACK to Homie-cf3a07e0
1458679257: Received SUBSCRIBE from Homie-cf3a07e0
1458679257: 	sensors/homie/cf3a07e0/relay/on/set (QoS 1)
1458679257: Homie-cf3a07e0 1 sensors/homie/cf3a07e0/relay/on/set
1458679257: Sending SUBACK to Homie-cf3a07e0
...
1458679258: OpenSSL Error: error:14094410:SSL routines:SSL3_READ_BYTES:sslv3 alert handshake failure
1458679258: Socket error on client Homie-cf3a07e0, disconnecting.

1458679258: New connection from 192.168.1.192 on port 9993.
1458679258: New client connected from 192.168.1.192 as Homie-cf3a07e0 (c1, k15).
1458679258: Sending CONNACK to Homie-cf3a07e0 (0, 0)
...
```

It's been running in TLS mode now for just over 7 minutes with only that one disconnection. I will keep it running like this for the time being.

jpmens, 2016-03-22:
That disconnect/reconnect in TLS mode is reproducible. I thought I'd add a _freeheap_ publish and show you what I see:

```
sensors/homie/cf3a07e0/$nodes light:cell,temperature:temperature,relay:relay
sensors/homie/cf3a07e0/$online true
sensors/homie/cf3a07e0/$name Livingroom bookshelf
sensors/homie/cf3a07e0/$localip 192.168.1.192
sensors/homie/cf3a07e0/$fwname h-sensor
sensors/homie/cf3a07e0/$fwversion 1.0.1
sensors/homie/cf3a07e0/temperature/unit c
sensors/homie/cf3a07e0/temperature/aaaaaaa Wemos-D1
sensors/homie/cf3a07e0/$signal 100
sensors/homie/cf3a07e0/temperature/degrees 22.81
sensors/homie/cf3a07e0/temperature/freeheap 14680
sensors/homie/cf3a07e0/light/cell 1
sensors/homie/cf3a07e0/temperature/degrees 22.81
sensors/homie/cf3a07e0/temperature/freeheap 14800
sensors/homie/cf3a07e0/temperature/degrees 22.81
sensors/homie/cf3a07e0/temperature/freeheap 14800
sensors/homie/cf3a07e0/$online false
sensors/homie/cf3a07e0/$nodes light:cell,temperature:temperature,relay:relay
sensors/homie/cf3a07e0/$online true
sensors/homie/cf3a07e0/$name Livingroom bookshelf
sensors/homie/cf3a07e0/$localip 192.168.1.192
sensors/homie/cf3a07e0/$fwname h-sensor
sensors/homie/cf3a07e0/$fwversion 1.0.1
sensors/homie/cf3a07e0/temperature/degrees 22.81
sensors/homie/cf3a07e0/temperature/freeheap 14624
sensors/homie/cf3a07e0/temperature/degrees 22.81
sensors/homie/cf3a07e0/temperature/freeheap 14624
sensors/homie/cf3a07e0/temperature/degrees 22.81
sensors/homie/cf3a07e0/temperature/freeheap 14176
sensors/homie/cf3a07e0/temperature/degrees 22.81
sensors/homie/cf3a07e0/temperature/freeheap 14624
sensors/homie/cf3a07e0/temperature/degrees 22.75
sensors/homie/cf3a07e0/temperature/freeheap 14624
```

Interestingly, after the reconnect, the topic `temperature/aaaaaaa` and `temperature/unit` are not published from the `setupHandler()`. What could that be due to?

marvinroger, 2016-03-22:
This is the intended behavior: the messages you're meant to publish from the setup handler are retained, so there is no need to send it again.

The issue does not seem to come from the heap, it seems stable... I've tried with an empty Homie sketch, and the heap available is always the same, but still it reconnects sometimes.

jpmens, 2016-03-23:
Good morning: device is still running and no reconnects during the night; the heap is stable.

The reconnect is a bit ugly, but I suppose we can live with that ;-)

marvinroger, 2016-03-23:
Good morning! This is really weird... There is no dynamic allocation between the first reconnect and the next ones, so why would the first fail? Anyway, good news it's more stable. :)
Maybe we can close this issue and open a new one about SSL stability.

jpmens, 2016-03-23:
Thank you very much for implementing `base_topic`! :+1:

closed 2016-03-23 by jpmens