homieiot/homie-esp8266 #45: Option to change publish path from "devices/device id" to "devices/name"

hoalex, 2016-03-22:
Hi there,

I think it would be great if Homie could optionally publish its MQTT messages under ```devices/name``` instead of ```devices/device id```. That way, multiple ESP8266s would be easier to distinguish, as one could simply name them -- for example -- "kitchen", "living room" and so on in configuration mode and then access them by that name via MQTT.
I guess that the current approach was chosen mainly to avoid naming conflicts, therefore I suggest using an optional statement in the ```setup()``` block à la:
```
Homie.useNameForPublish(true);
```

What do you think? :-)

jpmens, 2016-03-23:
`name` can contain a descriptive text such as `Light relay in living room`, which is great because it allows, say, a Web interface to display this clearly. As an avid hater of white space in topic names (and file system path names) I think this would lead to some confusion. (Of course people could suppress white space in name if they don't want it (`light-relay-in-living-room`).

I tend to think this calls for an additional setting (`sub_topic` ?) which ought to be in the configuration, as we did for `base_topic` in #44.

My 0.01c :smile:

marvinroger, 2016-03-23:
@jpmens This is what I wanted to do, an additional `device_id` field in the JSON configuration file. It would adhere to the following rules, as stated in [the convention](https://github.com/marvinroger/homie):

> Each device must have a unique device ID. This ID MAY be composed of lowercase letters from a to z, numbers from 0 to 9, and it MAY contain -, but MUST NOT start or end with a -.

@hoalex an `Homie.useNameForPublish(true);` method would not fit as Homie sketches are meant to be portable.

Does it sound good to both of you?

hoalex, 2016-03-23:
Sounds perfect :+1: Thanks for looking into it!

marvinroger, 2016-03-28:
This will land in 1.3.0! The new `device_id` field is at the top at the root of the JSON config.

closed 2016-03-28 by marvinroger

marvinroger, 2016-04-02:
Released!