homieiot/homie-esp8266 #47: Make exposed HomieNodes configurable on a per-device basis?

jpmens, 2016-03-23:
In #45 @marvinroger said something I find is important:

> Homie sketches are meant to be portable.

Assuming I have a few ESP8266 sensors which could basically all run the same Homie firmware, most of which have the same sensors / actuators on them, but, say, one or two of them have no _relay_. I would write the Homie sketch as follows:

```
HomieNode lightNode("light", "light");
HomieNode relayNode("relay", "relay");
HomieNode tempNode("temperature", "whatever");
```

When the Homie device _without_ the relay talks to me it would say "I have a relay!" which it doesn't. While this is just a detail, could we add an an element to the JSON configuration which basically says _hide these nodes for this device_? Something along the lines of

```json
{
  "hide_nodes" : [ "light", "relay" ]
}
```

I could then deploy the same Homie firmware to all devices and have them report / react to only what they actually have.

This may be overkill and not too easy to implement, so please feel free to quickly close this. :smile:

marvinroger, 2016-03-23:
By "portable", I mean a single firmware could be used by different people for their own same-hardware device. I did not think about what you're talking about here.

The idea is interesting, for an actuator (like a relay), the input handler would never be triggered, great. But for a sensor reporting values on a regular-basis, you would need to check in your own code if the node is enabled before sending a value.

It seems rather complicated to do. We need to focus on the stability of the framework, as you can see, adding a simple feature like the `base_topic` introduced a lot of heap issues. I won't close this, just in case, but I am not sure I will implement it. I am not comfortable with the idea of an all-in-one firmware. For my devices, I prefer a single purpose optimized firmware.

jpmens, 2016-03-23:
Fair points, and you're probably right, of course.
I'll close this proactively now.

closed 2016-03-23 by jpmens