homieiot/homie-esp8266 #48: Update wiki to mention limitations of configuration Web UI/app

AndrewDonaldson, 2016-03-28:
I have been unable to get the Web UI to connect. I now believe it only works from an Android device ( I can't prove this, I don't own one). I tried safari and chrome on iOS and safari on OSX with no luck.

If I'm correct, can you please update the wiki to indicate that the Web UI / app only works from Android?

marvinroger, 2016-03-28:
The Web UI is not platform dependant, it should work on every modern browser. However, it makes use of very recent APIs like the fetch API which maybe is not implemented on Apple devices. I will update the code to polyfill these recent features.

marvinroger, 2016-03-29:
What version of Chrome did you try?

marvinroger, 2016-03-29:
[Safari does not work, but Chrome >= 42 works](http://caniuse.com/#search=fetch). I still have added a polyfill just in case,

AndrewDonaldson, 2016-03-29:
I used Chrome on iOS, 49.0.2623.73.

marvinroger, 2016-03-30:
So you open Chrome 49, go to http://marvinroger.github.io/homie-esp8266 (no https), connect to the Wi-Fi network spawned by the ESP8266, and nothing happens?

AndrewDonaldson, 2016-03-30:
Trying it today, it's different on every browser. I suspect the polyfill update to the configurator changed things.

Prior to today, I loaded http://marvinroger.github.io/homie-esp8266 (no https), connect to the Wi-Fi network spawned by the ESP8266, and all clients sat at the "waiting for device" page.

Today, I loaded http://marvinroger.github.io/homie-esp8266 (no https or local zip download of it), connect to the Wi-Fi network spawned by the ESP8266, and all clients (OS X safari, iOS safari and iOS chrome) load the homie, let me progress through the web forms and then it fails to write configuration in the last step.

I have been working to compile the previous version of the configurator to recreate yesterdays result but my npm seems upset. I don't know any javascript.

Further note, so I stuck a USB wifi on my Linux box with chrome 47 go to http://marvinroger.github.io/homie-esp8266 (no https), connect to the Wi-Fi network spawned by the ESP8266 and it all works from there.

marvinroger, 2016-04-02:
There was an issue with the last step, could you please retry with the latest 1.3.0 release?

AndrewDonaldson, 2016-04-04:
Retried, same result.
Confirming, I used chrome 49 on iOS, went to http://marvinroger.github.io/homie-esp8266 (no https), switched to the ESP8266 wifi network, went through each web page and on the last one it says "There was an error while sending the configuration. Please retry". I did it twice.

Tried once with safari on iOS also, same result.

marvinroger, 2016-04-04:
Are you using v1.3.0 on the ESP8266? If yes, could you please send here the configuration you are using? Just change the sensitive data with random data. ;)

AndrewDonaldson, 2016-04-04:
Yes, I downloaded the latest homie-esp8266-master (library.properties says it is 1.3, I also deleted 1.2).
I can add that the serial monitor printed "Received CORS request for /config" and nothing else apart from wifi scans.

Config attached, this should be indicative of what I set in the web interface.

[ad-config-cleaned.json.zip](https://github.com/marvinroger/homie-esp8266/files/202582/ad-config-cleaned.json.zip)

marvinroger, 2016-04-13:
This is not normal, if the CORS request is triggered from your browser, then it should also, once the response is received, send the configuration to the device. It seems like this is an OS-related bug or something, I don't have issues on any Android phones, Windows and Linux computers...

sundariitm, 2016-07-27:
I am having the same problem. The Web UI doesnt connect, the App also behaves the same. For Web UI i tried http link  and local downloaded one. Chrome version: 51  & IE 8 .  
Also tried to use PUT application/json file by using a Chrome App but here it complains of JSON file being too large. 

need help to get this exciting stuff working please

marvinroger, 2016-07-27:
@sundariitm what version are you using?

sundariitm, 2016-07-28:
It is a NodeMCU (0.9) ESP-12 module as it reads in the Arduino.  I have used this module in earlier projects and is known working.

For the version of homie i took the latest 1.5.0 that  I got from Arduino library manager

marvinroger, 2016-07-28:
From the Arduino library manager? I did not know it was even available there! So you use the Arduino IDE, not PlatformIO?

sundariitm, 2016-07-28:
Right.. I am not familiar with PlatformIO :)

sundariitm, 2016-07-28:
One other observation.

As Android app configurator the WiFi name expected should start with Homie-xxxxxxx however my ESP module is spawning a WiFi AP with name ESP_DA817C to which i am able to successfully connect even without any password.  

See snapshot of my android phone screen

[image]
[image]

marvinroger, 2016-07-28:
I've never seen such a behavior... What esp8266/Arduino version are you using? 2.2.0? 2.3.0? Can you please provide your sketch, too?

sundariitm, 2016-07-28:
I dont know what really happened. I just reflashed the stable FW into NodeMcu and now the configuration is through.

thank you for follow-up...  the stuff you built is awesome and a good starting point for my project

marvinroger, 2016-09-03:
@sundariitm @AndrewDonaldson On the devices on which the request fail in the last step, could you please open the inspector (like F12 on Chrome), and in the Network tab you should have an OPTION request to `/config`. Could you please copy/paste the headers of the request? I probably identified the issue.

sundariitm, 2016-09-04:
the issue got fixed and i am unable to reproduce it now.. everything works
as expected.

AndrewDonaldson, 2016-09-04:
Upgraded to Homie 1.5, updated my program for the new API, tested etc. Then reset into Configuration mode and tested from OSX. It failed on the web config again.

I'm no web guy, so it took me a while to find the headers in Safari 9.1.3 (latest) on OSX. Screenshot attached, it didn't lend itself to cut and paste as far as I can see. Apologies if there was a better way to do this.

It wasn't obvious how to do this on iOS, likely the fault is the same as OSX though.

[image]

marvinroger, 2016-09-07:
Can you try replacing in `Homie/Strings.hpp` `PROGMEM_CONFIG_CORS` by:

```cpp
  const char PROGMEM_CONFIG_CORS[] PROGMEM = "HTTP/1.1 204 No Content\r\nAccess-Control-Allow-Origin: *\r\nAccess-Control-Allow-Methods: PUT\r\nAccess-Control-Allow-Headers: Content-Type, Origin, Referer, User-Agent\r\n\r\n";
```

If it still does not work, could you please also send a screenshot of the other `config` in the left panel? You see, there is the red `config — homie.config`, but there is another black, just above.

AndrewDonaldson, 2016-09-11:
It still didn't work. Screen shorts attached as requested. Thanks for continuing to look at this.

[image]
[image]

marvinroger, 2016-10-07:
I have no idea of what's going on... I changed the HTTP library for the new to-be-released configurator for the v2. Maybe it will help to solve this, although I am not convinced about that. This won't be a blocking issue for the v2 anymore though, as I don't know what I can do to solve this!

boneskull, 2017-03-02:
Current `master` seems to work fine with Safari v10.0.3 (12602.4.8).  Can't speak for v1.5.

marvinroger, 2017-06-01:
Does not happen on macOS Sierra with Chrome. Look like they got this fixed.

**For people who see this: there's a CORS bug affecting some Apple devices. Newest releases of Apple OSes fix this, but we cannot do anything to fix Apple issues. Please use another devices if you are affected.**

closed 2017-06-01 by marvinroger