homieiot/homie-esp8266 #49: Describe CLI method for uploading configuration

AndrewDonaldson, 2016-03-28:
I have been unable to upload a configuration with the Web UI / app and I don't own an android device.

I believe I should have been able to upload a configuration using curl, but so far I haven't had success. 

An example for doing this in the wiki would be appreciated. Then I could reduce the scope of what I am doing wrong.

I tried:
curl --upload-file config.json http://homie.config/config

which gives me {"success":false}, with a 400 error code.

github doesn't like me attaching the config file. I tried your complete example, simplified ones etc.

marvinroger, 2016-03-28:
As mentionned in the API, you must send the data as `application/json`:

`curl -X PUT http://homie.config/config -d @config.json --header "Content-Type: application/json"`

Open the serial monitor, and you will see logs with the reason your request has been denied.

rigorm, 2016-03-29:
If it helps, I use the "Chrome poster" extension under chrome to send the JSON info

I simply copy/pasted the JSON structure and changed what I needed then, click on "PUT".

just like the API page says: paste under http://homie.config/config

no need to upload a file :)

marvinroger, 2016-03-29:
Thanks for the tip @rigorm ;)

closed 2016-03-29 by marvinroger

AndrewDonaldson, 2016-03-29:
Thank you, that worked. I see you added it to the documentation also.

marvinroger, 2016-03-29:
You're welcome!

commit ceb1f70b (2016-03-29): Fix #49