homieiot/homie-esp8266 #50: Let the OTA server know which version was announced via MQTT

marvinroger, 2016-03-28:
As asked by @rschaten on #35

rschaten, 2016-03-28:
Thanks! :+1:

marvinroger, 2016-03-28:
The new version string is `da828fe0=sub2-relay@1.0.0->1.1.0`.

* `da828fe0`: the device ID
* `sub2-relay`: the firmware name
* `1.0.0`: the firmware version
* `1.1.0`: the OTA version announced via MQTT

marvinroger, 2016-03-28:
This will be released with 1.3.0 when the stabilty issues are resolved.

closed 2016-03-28 by marvinroger

jpmens, 2016-03-29:
Is there a particular reason you didn't opt for a single-character separator?

```
da828fe0=sub2-relay=1.0.0=1.1.0
```

That would make it a bit easier to _split_ on reception, wouldn't it?

marvinroger, 2016-03-29:
Well yes, it's way easier to split with a single char. I don't know why I did this. :disappointed_relieved:  Fixed 2481bdfa60b0c8f686c927cfefe47f240229590d

marvinroger, 2016-04-02:
Released. :)