homieiot/homie-esp8266 #53: Request: Add persistence to items

rigorm, 2016-03-31:
What I mean is, send the state of the items at a interval.

I'm seeing some weird actions with buttons on openHAB as, I would click them and they would revert back but the sensor would be in the asked state. so, by sending the states at a interval, the server would know the states.. (hope that i'm clear enough) :) # # @

marvinroger, 2016-03-31:
This has been discussed before, I can't remember where, but this is an openHAB issue. One of the benefits of MQTT is to be able to send retained messages, so we don't have to send them at an interval as you say.

rigorm, 2016-03-31:
Alright then.. I guess that the issue can be closed

closed 2016-03-31 by rigorm

marvinroger, 2016-03-31:
Sorry! I'm sure you understand that we cannot fix the issues of every controllers on the market. ;)