homieiot/homie-esp8266 #55: Advertise device through mDNS and DNS-SD

marvinroger, 2016-04-03:


marvinroger, 2016-04-19:
Finally there is no need to: the device already advertise itself through the MQTT. So this would be a complete waste of resources.

closed 2016-04-19 by marvinroger

rigorm, 2016-04-19:
Actually, I would put this one on hold.. devices are advertised on the MQTT but i'm not sure that they will be discovered as such. finding this via wifi would make it more effective.. MQTT is more of a transport method.. understand what i mean ? ;)

marvinroger, 2016-04-20:
Memory is precious, this is a waste of resources. Let me explain: if we make devices discoverable via mDNS, you can detect there is a device connected to the network and it's IP. Great. Then what? If you want to get other information about the device, or interact with it, you need to do it via MQTT anyway.

So simply, just subscribe to `devices/+/$online`, check if it equals to `true` and if yes, apply the regex `^devices\/([a-z0-9-]+)\/\$online$` to the topic, so you have the device ID in `$1` and you've just discovered all Homie devices. :smile:

rigorm, 2016-04-20:
I agree on that point.. (did not know about the + un a MQTT subscribe string). 
I was looking into binding making on openhab2 and looks like you need predefined "things" switch, sensors, relay etc.. So i'm so sure this would be a good thing for your framework as you "thing" can virtually be anything a use defines. 

Bottom line, this stays closed :)

qtronik, 2016-04-23:
Wow! Should put this trick in the readme file!

marvinroger, 2016-04-23:
Good point! But where? In the Homie convention, or in the docs of this repo?