homieiot/homie-esp8266 #56: Choose different subnet for configuration mode IP

mountaindude, 2016-04-06:
The 192.168.1.1 net is quite often used by SOHO routers. 
Placing a homie device in config mode on that same IP might easily cause conflicts with the user's cable modem/router/Internet gateway.

Maybe something like 192.168.50.1 (or something similarly infrequently used subnet)?

Or provide a way for the developer to define the desired subnet themselves?

mountaindude, 2016-04-06:
I guess one could also change the Homie source.. Looks like the 192.168.1.1 is defined in the src/Homie/Boot/BootConfig.cpp file (I am using Homie together with PlatformIO, btw)

mountaindude, 2016-04-06:
...then again, changing the access point's IP address to 192.168.50.1 breaks the config tool at http://marvinroger.github.io/homie-esp8266, it seems...

marvinroger, 2016-04-07:
The problem is changing the default `192.168.1.1` would be a breaking change. And `192.168.50.1` might also be used by someone else! So yes, maybe we should move the `apIp` variable to the `Constants.hpp`, so that it is easily customizable.

It should not break the config tool, it tries to reach `http://homie.config`, and as [you can see](https://github.com/marvinroger/homie-esp8266/blob/master/src/Homie/Boot/BootConfig.cpp#L49), the DNS server is tied to the `apIp`.

marvinroger, 2016-04-15:
Is it enough for you?

mountaindude, 2016-04-15:
Yeah it sounds like a good plan.

marvinroger, 2016-04-15:
It will land in v1.4.0. :)

closed 2016-04-15 by marvinroger

marvinroger, 2016-04-17:
Released.