homieiot/homie-esp8266 #57: Publishing of messages not working

mountaindude, 2016-04-07:
I don't Homie responding to ANY MQTT messages, which could be cause by

1. Incorrect MQTT topic being used by me. For the demo on https://github.com/marvinroger/homie-esp8266/blob/master/docs/2.-Getting-started.md, I understand the topic should be devices/003/light/on and the message should be true (assuming the device ID was set to 003 in the config).
2. Some issue in Homie, preventing it from getting the MQTT messages
3. Probably lots of other reasons too.

Also, after reporting the system messages to MQTT, the device sends a couple of "devices/003/$signal" messages, then goes offline with a "/devices/003/$online" = false message..

Would be good to rule out option 1. Can someone confirm that topic and message is correct, for the demo code referenced?

Using an ESP201 Sweet peas ESP8266 device, with latest PlatformIO (2.8.6) and Homie (1.3.0).

dholmen, 2016-04-07:
You should use topic devices/003/light/on/set => true/false

marvinroger, 2016-04-07:
Yes, to receive messages you need to send them on `devices/003/light/on/set`..

The `"/devices/003/$online" = false` is not normal, it means the device is disconnected from the broker... I'll investigate.

marvinroger, 2016-04-08:
By the way, what is your MQTT broker?

mountaindude, 2016-04-08:
@dholmen, ah that can explain it. Will test that topic. I read the docs so many times... but missed that final "set"

@marvinroger: Mosquitto 1.4.2-4 running on a Synology NAS.

marvinroger, 2016-04-08:
@mountaindude read https://github.com/marvinroger/homie :)

mountaindude, 2016-04-08:
@marvinroger: Promise - I *did* read that (excellent) page a ton of times, just that it was 1.30 am in the morning... should have known better. lol

mountaindude, 2016-04-08:
Btw, the issue with the device going offline could be due to the way the serial port works on this particular ESP8266 unit. Will try powering it from just a cell phone charger rather than a computer USB port.

marvinroger, 2016-04-13:
@mountaindude I can't reproduce this behavior... Was it a power issue?

marvinroger, 2016-06-02:
Closing this, feel free to reopen.

closed 2016-06-02 by marvinroger