homieiot/homie-esp8266 #58: Error compiling basic homie sample

rossdargan, 2016-04-09:
I'm using the following:

Arduino IDE 1.6.6
ESP8266 board 2.1.0
ArduinoJson 5.0.7
Bounce2 0.2.0
PubSubClient 2.4.0
Homie 1.3.0

My board is a NodeMCU V0.9

I can flash other programs to the board fine using the above.

I have tried the following

```
#include <Homie.h>
void setup() {
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

I also tried adding the following includes at the top as I didn't know if they were required

```
#include <ArduinoJson.h>
#include <PubSubClient.h>
#include <Bounce2.h>
```

I get the following error when I try and compile the app

```
C:\Users\live\Documents\Arduino\libraries\homie-esp8266\src\Homie\MqttClient.cpp: In member function 'void HomieInternals::MqttClientClass::initMqtt(bool)':

C:\Users\live\Documents\Arduino\libraries\homie-esp8266\src\Homie\MqttClient.cpp:27:148: error: no matching function for call to 'PubSubClient::setCallback(std::_Bind_helper<false, void (HomieInternals::MqttClientClass::*)(char*, unsigned char*, unsigned int), HomieInternals::MqttClientClass* const, const std::_Placeholder<1>&, const std::_Placeholder<2>&, const std::_Placeholder<3>&>::type)'

   this->_pubSubClient.setCallback(std::bind(&MqttClientClass::_callback, this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3));

                                                                                                                                                    ^

C:\Users\live\Documents\Arduino\libraries\homie-esp8266\src\Homie\MqttClient.cpp:27:148: note: candidate is:

In file included from C:\Users\live\Documents\Arduino\libraries\homie-esp8266\src\Homie\MqttClient.hpp:4:0,

                 from C:\Users\live\Documents\Arduino\libraries\homie-esp8266\src\Homie\MqttClient.cpp:1:

C:\Users\live\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:108:18: note: PubSubClient& PubSubClient::setCallback(void (*)(char*, uint8_t*, unsigned int))

    PubSubClient& setCallback(MQTT_CALLBACK_SIGNATURE);

                  ^

C:\Users\live\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:108:18: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (HomieInternals::MqttClientClass::*)(char*, unsigned char*, unsigned int), HomieInternals::MqttClientClass* const, const std::_Placeholder<1>&, const std::_Placeholder<2>&, const std::_Placeholder<3>&>::type {aka std::_Bind<std::_Mem_fn<void (HomieInternals::MqttClientClass::*)(char*, unsigned char*, unsigned int)>(HomieInternals::MqttClientClass*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>}' to 'void (*)(char*, uint8_t*, unsigned int) {aka void (*)(char*, unsigned char*, unsigned int)}'

exit status 1
Error compiling.
```

Sorry it's probably something really obvious, I just can't figure it out!

marvinroger, 2016-04-09:
Indeed: https://github.com/marvinroger/homie-esp8266/blob/master/docs/2.-Getting-started.md#1a-for-the-arduino-ide

You need PubSubClient >= 2.5.0 :)

rossdargan, 2016-04-10:
Doh, that was it (2.4.0 is the latest you will get from the arduino manage library stuff)

closed 2016-04-10 by rossdargan

jpmens, 2016-04-10:
@rossdargan Arduino IDE 1.6.8 has what you need: PubSubClient 2.6.0