homieiot/homie-esp8266 #60: mqtt.auth is not a boolean error when configuring with mqtt authentication ticked

rossdargan, 2016-04-10:
I'm getting the error "mqtt.auth is not a boolean" via serial when I send the config with "Use MQTT authentication" ticked (and my credentials entered).

rossdargan, 2016-04-10:
By the way same error using Chrome and Firefox on Android.

marvinroger, 2016-04-12:
Again, [check your dependencies](https://github.com/marvinroger/homie-esp8266/blob/master/docs/2.-Getting-started.md#1a-for-the-arduino-ide). ArduinoJson >= 5.0.8. There is a bug [I submitted to the ArduinoJson issue tracker](https://github.com/bblanchon/ArduinoJson/issues/214), that is fixed in 5.0.8.

rossdargan, 2016-04-12:
:( I'm really sorry - I did install all these libraries through the ide. I have no idea why they are all out of date. Is that usual?

marvinroger, 2016-04-12:
I have no idea, everything is up-to-date on my library manager!

marvinroger, 2016-04-12:
But yes as @jpmens said, maybe update your Arduino to 1.6.8?

rossdargan, 2016-04-12:
You are of course correct, sorry! 1.6.8 did allow me to update

closed 2016-04-12 by rossdargan