homieiot/homie-esp8266 #61: Can't upload config via spiffs

rossdargan, 2016-04-11:
This is the log from the upload. I placed the config.json file under data\ and data\homie\ because I wasn't sure where would work, neither did it though.

```
 [SPIFFS] data   : D:\GitHub\Arduino-Projects\HomieLounge\HomieLounge\data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/config.json

/homie/config.json

[SPIFFS] upload : C:\Users\live\AppData\Local\Temp\buildb84f8a28f2eaec1441ad2edf612d19ab.spiffs/HomieLounge.spiffs.bin
[SPIFFS] address: 0x100000
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : COM4
[SPIFFS] speed  : 115200

Uploading 3125248 bytes from to flash at 0x00100000
```

My config.json looks like this:

```
{
  "name": "Lounge Homie",
  "device_id": "lounge-homie",
  "wifi": {
    "ssid": "****",
    "password": "****"
  },
  "mqtt": {
    "host": "192.168.1.22",
    "port": 1883,
    "base_topic": "devices/",
    "auth": true,
    "username": "****",
    "password": "****",
    "ssl": false,    
  },
  "ota": {
    "enabled": false,    
  }
}
```

marvinroger, 2016-04-12:
Related to #60, I guess. The same check is performed on the `config.json` when sent over 
Serial or the API.

rossdargan, 2016-04-12:
You are of course correct, sorry!

closed 2016-04-12 by rossdargan