homieiot/homie-esp8266 #62: Trying to use the node input handler

rossdargan, 2016-04-12:
I am wanting to use the node input handler as I'm planning on controlling a set of switches. The property will relate to the switch, and the value will tell me if it is on or off.

My code looks like this:

```
HomieNode rcSwitchNode("rcSwitch", "switch", rcSwitchOnHandler);

bool rcSwitchOnHandler(String property, String value) {
  int houseCode = 1;
  int socketCode = property.toInt();

  controlStatusSwitch(houseCode,socketCode,value=="true");
  if (value == "true") {    
    Homie.setNodeProperty(rcSwitchNode, property, "true"); // Update the state of the light
    Serial.println("Switch is on");
  } else if (value == "false") {    
   Homie.setNodeProperty(rcSwitchNode, property, "false");
    Serial.println("Switch is off");
  } else {
    return false;
  }
  return true;
}

void setup() {  
  mySwitch.setProtocol(1);
  mySwitch.setRepeatTransmit(10);
  mySwitch.setPulseLength(307); // Bye Bye Standby
  mySwitch.enableTransmit(5);

  Homie.setFirmware("homie-switch", "1.0.0");
  Homie.registerNode(rcSwitchNode);
  Homie.setup();
}
```

Now I think my issue is that I'm not calling Homie.Subscribe("property",rcSwitchOnHandler) but I want to allow any property to be sent through really (well a number from 1->14).

This is the error homie gives me:

> Node rcSwitch not subscribed to 1
 
Thanks for this wonderful library!

marvinroger, 2016-04-13:
What about a 4th parameter in the `HomieNode` constructor, that if set to `true` would by default subscribe to every properties? So you would only have to: 

`HomieNode rcSwitchNode("rcSwitch", "switch", rcSwitchOnHandler, true);`

rossdargan, 2016-04-13:
yeah, that would work nicely!

marvinroger, 2016-04-13:
Great!

marvinroger, 2016-04-14:
Will land in v1.4.0. :)

closed 2016-04-14 by marvinroger

rossdargan, 2016-04-14:
Hi Marvin

I'm not a c dev, but I think the new changes could result in the property handler getting called twice. I've added a commend to the check in.

rossdargan, 2016-04-14:
In fact the https://github.com/marvinroger/homie-esp8266/blob/master/docs/3.-Advanced-usage.md suggests that the node input handler should handle every changed property of a specific node, but at the minute it doesn't. The same way the global input handler will not be called unless there is at least one subscription on that node.

An alternative could be to move https://github.com/marvinroger/homie-esp8266/commit/2bc6c8dcb9c7deafdd98ae69f2ae12241d1ba3b3#diff-ee62c9806128c2e2d545222f1f62952aR255 doewn below the global input handler, and the node property handler call? 

That way you wouldn't need the subscribe all stuff (and it would more closely match the docs at the minute)

marvinroger, 2016-04-14:
I am not sure I understand what you mean. It handle every changed property of a specific node, if this property is registered. In your case, it will be called since you will register the node to all properties.

marvinroger, 2016-04-14:
And the property handler [is called once](https://github.com/marvinroger/homie-esp8266/blob/master/src/Homie/Boot/BootNormal.cpp#L274), why would it be called twice?

rossdargan, 2016-04-14:
Had another look through, and I totally get it now! This is a great solution, thanks.

marvinroger, 2016-04-14:
You're welcome! :)

marvinroger, 2016-04-15:
@rossdargan before I release, can you test if it works for you? I added a `LedStrip` example for a concrete use case.

marvinroger, 2016-04-15:
Link of the archive : https://github.com/marvinroger/homie-esp8266/archive/7c4f6ba6a3eacf08cac6e3b949d1d7e7c4a8875a.zip

rossdargan, 2016-04-15:
Sure thing (It may be Saturday night now though I'm afraid!)

marvinroger, 2016-04-15:
No problem. :smile:

webguy16, 2016-04-15:
Hello, thanks for the great work on this, not OP, but i figured i would give this a shot (already have a sous vide cooker going with the homie library) and it looks to be working well. Here is the code i am using: 
https://github.com/webguy16/espCloner/blob/testing/espCloner_Homie/src/espCloner_Homie.ino

marvinroger, 2016-04-15:
Thanks a lot for your feedback! Glad to hear it is stable for you.

rossdargan, 2016-04-16:
Tested, and all working for me - fab! Here is my project https://github.com/rossdargan/Arduino-Projects/blob/master/HomieSwitch/HomieSwitch.ino

marvinroger, 2016-04-16:
Awesome, I guess it's ready for prime-time, then. :)

marvinroger, 2016-04-17:
Released!