homieiot/homie-esp8266 #65: Wish: Static configuration mode (e.g for devices with 512k Flash only)

euphi, 2016-04-16:
My ESP8266-ADC has 512k flash size only, so OTA is not possible.

This means I need to flash manually and reconfigure the device afterwards.

It would be nice to be able to hardcode Wifi-Settings.

As I'm using eclipse (and not the arduino IDE) there is also no tool to upload the config file manually to SPIFFS.

BTW: For me, the key feature of "homie-esp8266" is the easy API for handling mqtt messages. OTA/JSON-Configuration is a nice feature, but a little bit overengineered for my use cases.

marvinroger, 2016-04-16:
You don't have to reconfigure the device after a flash, as the SPIFFS is untouched when you flash the firmware. So you only have to configure it once. Are you sure your SPIFFS gets erased when reflashing?

euphi, 2016-04-17:
The config may have been deleted, because I have a jumper on GPIO0 to enter flash mode. I will try again with a changed (Homie-)reset pin.

euphi, 2016-04-23:
Ok, it seems it was caused by the GPIO0 as reset pin. Sorry.

marvinroger, 2016-04-24:
No problem. :)

closed 2016-04-24 by marvinroger