homieiot/homie-esp8266 #66: Is imroy pubsubclient more reliable?

jpmens, 2016-04-16:
Sorry to raise this question here, but I'm wondering whether somebody here knows if https://github.com/imroy/pubsubclient is maybe more reliable/stable than the orignal PubSubClient for ESP8266. I've seen some hiccups on ESP8266 and Homie (reported here earlier), and I'm wondering whether this library maybe fixes those?

marvinroger, 2016-04-16:
I also asked for feedback on the openHAB forum, maybe you saw it. This is something I definitely consider, and I started to work on it by abstracting MQTT functions into the [MqttClient class](https://github.com/marvinroger/homie-esp8266/blob/master/src/Homie/MqttClient.cpp), so it will be a piece of cake to make the switch.

marvinroger, 2016-04-16:
I will try to benchmark the two libraries, to monitor the times it takes to connect / subscribe / publish, how many times it reconnects, and how much memory it uses.

jpmens, 2016-05-03:
FWIW, I've done a few tests with a simple sketch that subscribes to a single topic, prints the payload string to Serial: there's not much difference between the two libraries in terms of stability. I can fire off about 100 messages (qos=0, qos=1) and they _tend_ to arrive without resetting the device. Anything above that number quickly crashes the module.

On the other hand, if I attempt to publish what the device received (i.e. play "echo" server), things quickly fall apart: while PubSubClient manages to echo two or three messages, the _imroy_ version just quietly dies.

This is meant as a heads-up, that it's probably not worth your while to pursue the modularization, so please close this asap. :-)

marvinroger, 2016-05-03:
Alright, thanks for testing! I will try to learn more about the MQTT network protocol, and attempt to implement something with the [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP) library. As it is non-blocking, it might handle a lot more workload and therefore be more stable. However, there is not support for SSL for now (https://github.com/me-no-dev/ESPAsyncTCP/issues/9).

jpmens, 2016-05-03:
Well, to be honest, within a closed environment, TLS is probably overrated anyway. 😆 

/me _ducks_

jpmens, 2016-05-03:
@marvinroger You should speak to @ckrey about this: he's our MQTT protocol guru. Honestly: he built OwnTracks for iOS and has done tons of MQTT protocol tests with different brokers. I'll ask him to create a repo to start working on this with you.

jpmens, 2016-05-03:
https://github.com/ckrey/mqtt-esp

marvinroger, 2016-05-03:
I can't work on this right now, I have my exams soons so I'll have to slow down. :confused:

Jorgen-VikingGod, 2016-05-04:
I had some issues in the past with pubsubclient and 10+ retained messages.
After connection all the messages are bombed into my ESP8266 and forces sometimes a crash.
To avoid this, I added a ring buffer and push all arrived messages (ok maybe not all ;) )
In my loop I process/pop all stored messages.
This is working really stable for me by the way

marvinroger, 2016-05-11:
Here is the asynchronous library I've been working on this week-end: [marvinroger/async-mqtt-client](https://github.com/marvinroger/async-mqtt-client). It seems pretty stable, but some exceptions seem to appear on disconnect... It would help a lot if I can have feedback on it. :smiley:

marvinroger, 2016-06-02:
AsyncMqttClient will be used, see https://github.com/marvinroger/homie-esp8266/issues/87

closed 2016-06-02 by marvinroger