homieiot/homie-esp8266 #67: V1.4.1 : Error when compiling door example

TommySharpNZ, 2016-04-19:
I seem to be having an issue when trying to compile the door example with 1.4.1...

Anyone able to comment on this error message?

C:\Users\Tommy\Dropbox\Arduino\libraries\homie-esp8266-master\src\Homie\Helpers.cpp: In static member function 'static HomieInternals::MdnsQueryResult HomieInternals::Helpers::mdnsQuery(const char*)':

C:\Users\Tommy\Dropbox\Arduino\libraries\homie-esp8266-master\src\Homie\Helpers.cpp:21:16: error: 'class MDNSResponder' has no member named 'queryService'

   int n = MDNS.queryService(service, "tcp");

C:\Users\Tommy\Dropbox\Arduino\libraries\homie-esp8266-master\src\Homie\Helpers.cpp:26:22: error: 'class MDNSResponder' has no member named 'IP'

     result.ip = MDNS.IP(0);

C:\Users\Tommy\Dropbox\Arduino\libraries\homie-esp8266-master\src\Homie\Helpers.cpp:27:24: error: 'class MDNSResponder' has no member named 'port'

     result.port = MDNS.port(0);

exit status 1
Error compiling.

marvinroger, 2016-04-19:
No, you are using the git development version. Released versions are available here : https://github.com/marvinroger/homie-esp8266/releases

marvinroger, 2016-04-19:
And if you want to compile the git version, you'll need Arduino for ESP8266 >= 2.2.0 which was releases yesterday.

marvinroger, 2016-04-20:
Feel free to reopen if needed!

closed 2016-04-20 by marvinroger