homieiot/homie-esp8266 #68: MAX_REGISTERED_NODES_COUNT too low

rschaten, 2016-04-19:
Looks as if I just overloaded the library... :-)

I built a homie device with several sensors, and it failed on me. Nothing useful (for me) in the debug output, so I traced the problem down to a call of Homie.registerNode(). I can register up to five nodes, but not a sixth one. After a look at your library, I found the limit of 5 hard coded in `Homie/Limits.hpp`. After setting this to 10, everything works fine for me.

Is it possible to make that limit configurable? Or would you do me a favour and generally increase the limit? At the moment, this one device needs six nodes, but I plan on increasing that to at least eight...

By the way, the implementation in `Homie.cpp` looks as if I should have gotten a warning about this problem ('the max registered nodes count has been reached'). That didn't happen. It died after 'Sending initial information...'.

```
Attempting to connect to MQTT...
Connected
Sending initial information... 
Exception (28):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000000a5 depc=0x00000000
```

webguy16, 2016-04-20:
I experienced the the no-debug message and resetting after 'Sending...' issue as well. Instead of increasing node count (relay1 node, r2 node) i worked it a different way (relay1 = property of node Relay). That being said, a  few things could be done. A proper debug message, making the node count configurable or increasing it would be helpful if stability/memory allows. Also, I might have missed it, but putting putting a note or two somewhere in the docs about the limit probably wouldn't be amiss either. Finally, thanks again for the awesome work you do on this.

marvinroger, 2016-04-20:
The warning should have been shown when calling `Homie.setup()` and then fail, so there must be a problem with the check condition.

The reason why there are hard limits is that I wanted to avoid most of the dynamic allocations, so everything is allocated on the stack and not on the heap. There are no ways to change the limits in the code in this (actuallcontext y this might be doable with templates, I'll investigate, but this would be a breaking change).

That's why these limits are in a unique header file, so you could change the constants easily. 5 seems like a reasonable default, I personally don't have any devices that have more than 3 nodes. This is I think a good compromise between used memory and flexibility. 

That being said, dynamic allocation might not be an issue here: the memory is only allocated once, so the memory fragmentation would be very limited. And it may even save memory, as if you are only using 1 node, 5 are actually allocated in the current code. 

Actually, there is already a section in the docs about this in the troubleshooting. But yes, the warning is not shown, this is error-prone.

rschaten, 2016-04-20:
I totally understand the reasons for static memory allocation, and I confess: having more than 5 nodes seems to be a bit excessive. But in this special case, all nodes really are of different types. Collecting them seems to be wrong.

But on the other hand: reading the Homie convention really looks as if a device with for example eight relays and eight LEDs should have 16 nodes. I don't have one, but I guess it's not too hard to imagine a use case for this. :-)

Thank you for considering this as an enhancement. If I was a better C++ developer, I would try making it dynamic myself and send you a pull request. But to be honest: I'm not. :-(

For the moment, I can live with setting the value in `Homie/Limits.hpp`. I just have to remember editing that file after each update...

marvinroger, 2016-04-20:
> reading the Homie convention really looks as if a device with for example eight relays and eight LEDs should have 16 nodes. I don't have one, but I guess it's not too hard to imagine a use case for this. :-)

Well, actually you can do whatever you want, as long as it fits to your requirements.
If you have 8 independant LED then yes, you can use 8 nodes. But if you have a LED matrix for example, I would use a single node. It really depends and there are no clear indications about this in the docs so that you can do what you think is best for you.

I'll remove the limits of the number of nodes and subscription for each node. ;)

euphi, 2016-05-16:
The error message is not shown on the serial, because Serial works asynchronous.
So you should add a Serial.flush() before the call to abort():

    void HomieClass::registerNode(HomieNode& node) {
      this->_checkBeforeSetup(F("registerNode"));
      if (this->_interface.registeredNodesCount > MAX_REGISTERED_NODES_COUNT) {
        Serial.println(F("✖ register(): the max registered nodes count has been reached"));
        Serial.flush();
        abort();
      }
    
      this->_interface.registeredNodes[this->_interface.registeredNodesCount++] = &node;
    }

marvinroger, 2016-05-16:
I thought Serial was sync! Thanks for information. :)

marvinroger, 2016-06-02:
#96 brings autoregistration and removes the limitation of max number of nodes, without using dynamic memory. Will land in 2.0.0. :)

closed 2016-06-02 by marvinroger

marvinroger, 2016-06-19:
Max number of subscriptions per node is now also unlimited. :+1: