homieiot/homie-esp8266 #69: cannot configure device

RREE, 2016-04-22:
Hi,
the project looks quite promising. Unfortunately I'm fighting issues nobody seems to have.

I can build and upload the Homie code to the device. I actually have two different devices, a Wemos D1 mini and a strange part called "ESP8266 serial WIFI Witty cloud"

1. I cannot find any Android app. The references page just says that no app is available and I should check later on

2. On my Android phone I'm connected to my standard WiFi net and I open your configuration page http://marvinroger.github.io/homie-esp8266. I then connecto to the Homie AP, type in my configuration and send it to the AP. As a response I get the error message "There was an error while sending the configuration. Please retry." In serieal output it tells me:
Received CORS request for /config
Received config request
... name is not a string
Triggering Wi-Fi scan...

3. I wrote my config.json manually and placed it in <sketch>/data/homie/config.json and started the upload in Arduino IDE. I takes quite some time and then responds:
Arduino: 1.6.8 (Windows 8.1), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

[SPIFFS] data   : C:\Users\Rolf\Documents\Arduino\homie_test1\data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/homie/config.json

[SPIFFS] upload : C:\Users\Rolf\AppData\Local\Temp\build4c3e170adfc69896b7d4f027890c0e0d.spiffs/homie_test1.spiffs.bin
[SPIFFS] address: 0x100000
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : COM5
[SPIFFS] speed  : 921600

Uploading 3125248 bytes from to flash at 0x00100000

...................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
error: espcomm_upload_mem failed
SPIFFS Upload failed!

DavidStacer, 2016-04-23:
Once the D1-mini is in WiFi AP mode, connect the PC wireless to it. Then I use the curl command to download the json configuration file. This uses the Homie configuration API

put the following in a file download.cmd in the same directory where the json file.
curl -X PUT http://homie.config/config --header "Content-Type: application/json" -d @test.json
pause

It will be interesting to see if you have problems with the D1-Mini going  back into config mode when you plug it into the USB of your computer.

Is it the original or the newer V2 versions?

marvinroger, 2016-04-23:
@RREE 

1. There is currently a problem with the Adobe PhoneGap build service, I cannot do anything on my side to fix the Android app availability.

2. The `name is not a string error is weird`... What do you fill as a name in the UI?

3. This is an Arduino for ESP8266 issue, the uploading to the SPIFFS fails. Fill an issue there. :)

marvinroger, 2016-04-23:
@RREE I identified the `name is not a string error` bug. I'll fix it soon!

marvinroger, 2016-04-23:
Will land in the next release.

closed 2016-04-23 by marvinroger

marvinroger, 2016-04-23:
Released!

RREE, 2016-04-24:
Thanks Marvin,
setting the config file via my mobile phone works now!

marvinroger, 2016-04-24:
Awesome! Unfortunately I can't do anything for the two other problems.

marvinroger, 2016-04-24:
The Android app is back up. :) It seems bigger than before however, but it should be good to go.

commit 8fb6d929 (2016-04-23): Fix #69