homieiot/homie-esp8266 #71: wemos d1-mini: cannot get in configuration mode

gtom, 2016-04-23:
Hi,
I did a misconfiguration in mqtt credentials, now my d1-mini is trying endless to connect to wrong server. I have a reset button on d1-mini, but after pressing it for 5 seconds, it always reboots into normal mode and tries to connect again. 
Top get into configuration mode, I have to press FLASH button, which doesn't exist on d1-mini. What can I do to get into configuration mode? Any way to reset the eeprom-file-system during flashing homie again?
Thanks in advance
Tom

marvinroger, 2016-04-23:
Just format the file system once:

```c++
#include "FS.h"

void setup() {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Mounting FS...");

  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount file system");
    return;
  }

  SPIFFS.format();
  Serial.println("Done.");
}

void loop() { }
```

If you want, you can also change the reset button pin (you might add an external button): https://github.com/marvinroger/homie-esp8266/blob/master/docs/3.-Advanced-usage.md#reset

TommySharpNZ, 2016-04-24:
It would be great to work out how to make the onboard button work on the Wemos D1 mini so that we can press it for 5 seconds to get into configuration mode.

DavidStacer, 2016-04-25:
The button on the D1-mini is tied directly to reset on the Esp8266 so I don't think its possible to capture that.  The Wemos button Shield ties pin D3 GPIO0 to GND.  If you don't have the button shield, you could emulate that with a simple jumper from D3 on the D1-mini to GND for 5 seconds.

The Wemos schematics can be found [here](https://github.com/wemos/wemos.github.io/blob/master/Products/d1_mini.md)

TommySharpNZ, 2016-05-21:
What do we need to do in our sketches to make D3 on the D1-mini become the reset button?

DavidStacer, 2016-05-21:
D3 on the D1-mini is already connected to GPIO0 on the ESP-12 and GPIO0 is the default for Homie, so you shouldn't have to do anything in the code. Just hold D3 to GND for 5+ seconds.

I did run into a problem where the D1 always wanted to go into config mode. [documented here](https://github.com/marvinroger/homie-esp8266/issues/64#issuecomment-210844811) .

marvinroger, 2016-06-02:
Sorry guys, as @DavidStacer explained, the D1 mini button is tied to the reset, so there is no way to intercept the signal.

closed 2016-06-02 by marvinroger