homieiot/homie-esp8266 #73: platformio: Version 1.5 does not compile due to old ESP8266mDNS.h version

euphi, 2016-04-24:
I updated my platformio lib installation, so Homie was updated to 1.5.

Now there a various compile errors, because Homie makes use of some mDNSResolver members that are not available in the latest (quite old) ESP8266 libs in platformio:

`.pioenvs/esp12e/Homie_ID555/Homie/Helpers.cpp:21:16: error: 'class MDNSResponder' has no member named 'queryService'
int n = MDNS.queryService(service, "tcp");
^
.pioenvs/esp12e/Homie_ID555/Homie/Helpers.cpp:26:22: error: 'class MDNSResponder' has no member named 'IP'
result.ip = MDNS.IP(0);
^
.pioenvs/esp12e/Homie_ID555/Homie/Helpers.cpp:27:24: error: 'class MDNSResponder' has no member named 'port'
result.port = MDNS.port(0);`

I also updated my complete platformio installation - everything is up-to-date, the compile error persists.

ESP8266mDNS.h  is part of the espressif SDK and it seems to be quite old in platformio.

marvinroger, 2016-04-24:
It's not part of the espressif SDK, it's part of the Arduino for ESP8266. The ability to query services has been added in Arduino for ESP8266 2.2.0, which is not yet updated in PlatformIO. You have to wait for https://github.com/platformio/platformio/issues/627 to be fixed.

bertnic, 2016-04-28:
i see there was a patch which i applied. still doesnt work though

marvinroger, 2016-04-28:
Update PlatformIO to 2.9.0, it should fix the issue.

bertnic, 2016-04-29:
fixed. Thanks a lot Marvin

marvinroger, 2016-04-29:
You're welcome!

closed 2016-04-29 by marvinroger