homieiot/homie-esp8266 #75: Some WiFi passwords characters are not valid

DavidStacer, 2016-04-25:
I have a very complex WiFi password and some of the characters in it caused Homie trouble.

Testing shows that:
% &  " ' + _      Cause issues. Some failed in the config loading, some never connected to wifi.
~!@#$^*()-={}[]:;<>?,./`Aa1234567890    All work.

This [Stackoverflow link ](http://stackoverflow.com/questions/4919889/is-there-a-standard-that-defines-what-is-a-valid-ssid-and-password) says any character should work for WiFi, so possibly its  a Web UI or json issue.

I tried in a native Arduino IDE 
const char* password = "~!@#$^*()-={}[]:;<>?,./`Aa1234567890%&+_'\"";  and that password worked. I had to escape the " with \"  .

I'm going to setup a separate IOT WiFi network so its not an issue for me but I wanted to document the characters that you can't use in your WiFi password and maybe save someone else some debug time.

List that won't work spelled out with hex
  percent  0x25
  ampersand 0x26
  double quote 0x22
  single quote 0x2C
  plus sign 0x2B
  underscore 0x5F

marvinroger, 2016-04-25:
Thanks for the investigation. This is indeed problematic! I'll do some testing.

marvinroger, 2016-06-19:
I just tried the `abcdef%&"'+_` password, and it works fine... I am using the git version, that needs the `2.3.0-rc2` version of esp8266/Arduino, so maybe some things got resolved there. Anyway, when Homie for ESP8266 v2.0.0 will be out, this should not be a problem anymore.

closed 2016-08-16 by marvinroger