homieiot/homie-esp8266 #77: Deep sleep - mqtt convention

piotrwachowiak, 2016-04-27:
Hi Marvin, thanks for the awesome work You did. This lib is perfect. 
For some of the sensors I use deep sleep mode, did You think about implementing dsleep procedure at library level ? Right now, I am doing this checking  isHomieReadyToOperate and deepSleep, however more graceful (like closing mqtt session) method will be appreciated. Of course it is not a problem, just an opportunity for improvement.

Thanks again !

Piotr

marvinroger, 2016-04-27:
Glad you like it! Thanks!

The library only implements configuration and network handling, and I would like to keep it as simple. The only requirement for you is to be able to gracefully disconnect from the MQTT broker? Then what about a method `Homie.disconnect()` that would disconnect the MQTT connection?

piotrwachowiak, 2016-04-27:
Thanks, this should do the job !
I was also thinking about modifying a bit MQTT convention to add state "sleep" and maybe properties to control sleep from MQTT, like $dsleep and $dsleep_time. 

Thanks a lot, I will spend more time learning details of Your code, till now I was trying to quick configure as many sensors as possible, modifying examples.

best regards,

Piotr

hoalex, 2016-05-12:
Any update on this? :-) I would like to use one of my nodes with batteries, so obviously deep sleep is a must to get more than a few hours/days of uptime. I think you're right in that Homie itself should only handle network-specific stuff, so I would appreciate the above-mentioned ``Homie.disconnect()`` method to gracefully disconnect from the MQTT broker before sending the node to sleep.

Thanks and regards
Alex

marvinroger, 2016-05-12:
Sorry, I have a lot of work these days! So I won't release a new version before quite a while.

For the time being, you can add to `Homie.cpp`, at the end:

```c++
void HomieClass::disconnect() {
  if (!this->isReadyToOperate()) {
    this->_logger.logln(F("✖ disconnect(): impossible now"));
    return;
  }

  this->_mqttClient.disconnect();
}
```

and in `Homie.hpp`:

```c++
// after bool setNodeProperty(const HomieNode& node, const char* property, const char* value, bool retained = true);
void disconnect();
```

It should do the trick. :)

Nzbuu, 2016-06-20:
As far as I can tell, MQTT has no support for sleep states. If you need a proper sleep state, then you should use the MQTT-SN protocol instead (See paragraph 6 of section 3 and section 6.14: http://mqtt.org/new/wp-content/uploads/2009/06/MQTT-SN_spec_v1.2.pdf).

I think that, for pure MQTT, you would have to disconnect and then retrieve any QOS >= 1 or retained messages when you reconnect.

marvinroger, 2016-06-26:
Yes @Nzbuu, this is the best thing to do.

As I said earlier, Homie and Homie for ESP8266 won't handle deep sleep. You can however DIY, with a special `system` node for example.

I just implemented a `Homie.disconnectMqtt()` method. Just intercept the `HOMIE_MQTT_DISCONNECTED` to know when you're actually disconnected cleanly from the broker, and then deep sleep for as long as you want. As a side note, I also made some improvements to improve the Wi-Fi connection time. In my tests, the total connection time (Wi-Fi and MQTT) went from 4 seconds to 1 second with the latest asynchronous code. If you want a temperature sensor with deep sleep, it means you will last 4 times more on a battery.

This will land in v2.0.0, no ETA however! Hope it suits you.

marvinroger, 2016-08-16:
e7124f28c0e47c1b96962732793d18e1e97dc8d5 adds access to the raw underlying MQTT client, so you can do any MQTT stuff you want (like disconnecting). Closing the issue, I'll let you know when the v2.0.0 is released. :)

closed 2016-08-16 by marvinroger