homieiot/homie-esp8266 #78: Two relays handling on single node

piotrwachowiak, 2016-04-27:
Hi Marvin,
 is there any way of passing additional arguments to handler in Homie.subscribe() method ? I am having troubles registering two relays on one node, do I have to create two separate handlers to pass different pin number ? Like:

HomieNode relayNode1("room", "relay1");
HomieNode relayNode2("room", "relay2");

[...]

  relayNode1.subscribe("on", lightOnHandler1);
  relayNode2.subscribe("on", lightOnHandler2);

or maybe it is possible to have node_id and/or type to be passed to handler ? 

Sorry if this question is not clear for You or the answer is simple, I am getting lost in object-oriented programming.

Thanks in advance,

Piotr

marvinroger, 2016-04-27:
I suggest you to take a look at the [LedStrip example](https://github.com/marvinroger/homie-esp8266/blob/master/examples/LedStrip/LedStrip.ino). It does exactly what you want, but with LED instead of relays - which is exactly the same, actually -. Don't hesitate to ask if you don't understand. :)

piotrwachowiak, 2016-04-27:
Thanks for quick reply ! I was trying to modify example from docs - Getting Started and LightOnOff example. Now it's clear for me, at least now I think so :) 

I have to go deep into detail, because I think I not fully understand node/property, registration and subscription. I will test some scenarios and probably everything will be clear. 

Thanks for help !

Piotr

closed 2016-06-02 by marvinroger