homieiot/homie-esp8266 #79: Querying mDNS causes "Exception (28)" reset loop.

webguy16, 2016-04-28:
Hi, im not really sure what im doing wrong here. I'm trying to use the mDNS service for setting up the mqtt and ota servers on the homie device.
Homie library: 1.5.0
Arduino: 1.6.8
esp8266: 2.2.0-rc1

node-red discovery node: 
service "_http._tcp" 
port "1883" 
txtRecord:""
name: "mqtt"

Triggering HOMIE_WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Querying mDNS service mqtt

Exception (28):
epc1=0x40216ac8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000                                                                             000

ctx: cont
sp: 3fff2fe0 end: 3fff32f0 offset: 01a0

>>>stack>>>
3fff3180:  4021004b 00000030 3fff20d0 40210053
3fff3190:  74716d5f 00000074 000001da 4020ffee
3fff31a0:  00000009 00000340 00000000 61636f6c
3fff31b0:  745f006c 3f007063 00000001 00000005
3fff31c0:  00000004 00000005 005f0000 40216ea4
3fff31d0:  3ffe8d74 00000003 3fff1cf4 3fff22bc
3fff31e0:  3fff1c14 00000004 3fff2200 3fff22bc
3fff31f0:  3fff1c14 3fff20d0 3fff3240 40208c90
3fff3200:  3fff1cf1 00000015 3fff2200 40214084
3fff3210:  3fff1cf1 0000075b 3fff2200 402140f8
3fff3220:  3ffeff40 3fff1bfc 3fff1174 4020bb17
3fff3230:  00000001 0018a95a 4020fca8 3fff1c04
3fff3240:  00000000 3ffe9af0 00000000 4020fc8a
3fff3250:  40207a20 00000023 3fff2200 40214084
3fff3260:  3fff2070 3fff1194 3fff2200 3fff22bc
3fff3270:  3fff1c04 3e4ccccd 3fff1c00 3fff22bc
3fff3280:  3fff2070 3fff1194 3fff1174 4020c5ed
3fff3290:  00000001 3fff11d0 3fff1174 4020c34a
3fff32a0:  3fff1c0c 3fff1bfc 3fff116c 3fff22bc
3fff32b0:  3fffdad0 00000000 3fff22b4 402165f2
3fff32c0:  00000000 00000000 00000001 40206eac
3fff32d0:  3fffdad0 00000000 3fff22b4 40215654
3fff32e0:  feefeffe feefeffe 3fff22d0 40100718
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x0f
csum 0x0f
~ld

marvinroger, 2016-04-28:
I'll investigate, thanks for the report!

rigorm, 2016-04-28:
should'nt that be  --> service "_mqtt._tcp"  and not  "_http._tcp"  ?

webguy16, 2016-04-29:
> should'nt that be --> service "_mqtt._tcp" and not "_http._tcp" ?

I'm really not sure. I just tried it with the same results though. Also, i could be wrong but It's my understanding i should have gotten an error message ("✖ Service not found") instead of the reset loop either way.

marvinroger, 2016-05-12:
Can you test with @unaiur's fix (#81)?

marvinroger, 2016-06-02:
Closing this as this is fixed upstream, and as @webguy16 seems inactive.

closed 2016-06-02 by marvinroger