homieiot/homie-esp8266 #80: provide a mechanise for one device to *listen* to another devices publishes

sumnerboy12, 2016-04-30:
Example - I have a Wemos D1 with the DHT22 shield, publishing temperature and humidity values to separate nodes. I would like to configure another Wemos with the OLED shield to display the values of these sensors. The idea being the sensor lives in some hard-to-reach place, and the display is in my living room for easy viewing.

I can't see a way to do this with Homie currently - unless I am missing something? I can obviously write a custom (non-Homie) sketch which subscribes to those MQTT topics and displays the values, but I thought this would be a good feature to add to homie.

Let me know your thoughts. I am guessing some sort of special subscribe handler which takes the device id, node id and property name, and a handler. Just a wrapper around an MQTT subscribtion which makes it easy to listen to other homie nodes.

jpmens, 2016-04-30:
Depending on the number of subscriptions thusly set up, I'd fear a bit of instability on the _subscribing_ device; my experiments with PubSubClient have shown that a few messages in a couple of seconds can cause the device to reset / reboot.

sumnerboy12, 2016-04-30:
Understood, but I still think it would be a useful feature - this remote display would be quite useful but currently can't be done within the Homie f/w. If someone subscribes to 45 different feeds, updating every 2s then be it upon their head!

marvinroger, 2016-05-02:
This would indeed be useful. However, as @jpmens said, there are issues when subscribing to more than 5 topics (https://github.com/knolleary/pubsubclient/issues/98). This can be fixed by adding a `loop()` call in between subscribe call. But there is another issue with SSL enabled, we can't subscribe to more than 3 topics (https://github.com/knolleary/pubsubclient/issues/141). Homie already subscribes to node properties, to `$ota` and to `$reset`, so I am afraid what you're asking won't be doable.

jpmens, 2016-05-02:
I wonder whether the [other pubsubclient](https://github.com/marvinroger/homie-esp8266/issues/66) would mitigate these issues ...

sumnerboy12, 2016-05-03:
Looks like I might have to use [mqttwarn](http://github.com/jpmens/mqttwarn) to pipe the messages from one node to the other. Not very elegant but it should work. 

Is there any reason you don't just subscribe to `basetopic/devicename/#`? This would only require a single subscription leaving room for a couple of user-defined subscribes.

jpmens, 2016-05-03:
@sumnerboy12 the publishes _from_ the device would come right back and probably kill it. That, at least is what my tests with PubSubClient (outside of Homie) have shown, but I'm very confident @marvinroger has something more intelligent to say about this. :-)

sumnerboy12, 2016-05-03:
good point! will leave this to the experts then.

closed 2016-05-03 by sumnerboy12

marvinroger, 2016-05-03:
This is is what we did first, but we [had to revert this](https://github.com/marvinroger/homie-esp8266/commit/d7ea186c1fb252c932bfb7c324882a759c7740d4), because as @jpmens [said](https://github.com/marvinroger/homie-esp8266/issues/46#issuecomment-201848926), the device was killing itself.