homieiot/homie-esp8266 #82: ESP07 doesn't boot into configuration mode

webworxshop, 2016-05-11:
Hi,

I'm building a simple Homie app with platformio for the esp07. Weirdly, the device seems to be unable to boot into configuration mode. There is also nothing intelligible logged to the serial interface. I've verified with a simple Hello World program which works just fine.

The even weirder thing is that when I set the board type to esp01, Homie appears to boot into config mode just fine, however I still get no intelligible log.

The platformio version is 2.9.1. Not sure of the versions or other software and libs, platformio doesn't seem to expose this information.

Please let me know if there are any more details I can provide.

marvinroger, 2016-05-11:
Have you tried [this](marvinroger.viewdocs.io/homie-esp8266~v1.5.0/9.-Troubleshooting#user-content-1-i-see-some-garbage-on-the-serial-monitor)? Is the log better?

webworxshop, 2016-05-12:
Your link was broken, but I managed to find the right place. That does fix the logging issue.

I'll re-try the build for ESP07 later and let you know what the log says.

marvinroger, 2016-05-12:
Sorry for the link, GitHub seems to encode the `~` in a weird way.

Thanks!

webworxshop, 2016-05-12:
No worries.

As an update, I get no log output at all when building for the ESP07 target.

I guess this could be a lower level platformio or Arduino issue, but I'm not sure where to look to debug further.

rigorm, 2016-05-12:
To give some upside to your problem, the ESP-07 do work. I have one on my test bench along side a ESP-12E and both are great with the Homie framework.

Did you try with the Arduino IDE instead of PlatformIO ? I use the "Generic ESP8266" when selecting the platform target and use a 512k/128k as a the flash size. (Actually, I use 1M/64K as a flash size but that is what i found I had on my esp-07, could be different for you)

marvinroger, 2016-06-19:
Closing this as there are no answers from the OP. Feel free to reopen.

closed 2016-06-19 by marvinroger

webworxshop, 2016-07-16:
Sorry for the lack of updates on this. Been busy with other things and haven't managed to get back to it. Looks like this is most likely a platformio issue, so OK to close here.