homieiot/homie-esp8266 #85: Configuration Site http://marvinroger.github.io/homie-esp8266/ broken!?

euphi, 2016-05-14:
I cannot configure my device using http://marvinroger.github.io/homie-esp8266/ . Whenever I connect to my ESP8266 the page shows the configured nodes, but it is not possible to move to the next page (The small icon on top changes to next, but browser still shows the info page.

`
Triggering Wi-Fi scan...
✔ Wi-Fi scan completed
Triggering Wi-Fi scan...
✔ Wi-Fi scan completed
Triggering Wi-Fi scan...
✔ Wi-Fi scan completed
Received heart request
Received device info request
`

marvinroger, 2016-05-14:
You're right, I tried to reduce the size of configuration UI to be able to embed it into the device itself, and I broke everything!

marvinroger, 2016-05-14:
Can you try again?

euphi, 2016-05-14:
Yes, works fine now. Many thanks!

marvinroger, 2016-05-14:
You're welcome!

commit fd70a998 (2016-05-14): Fix #85
closed 2016-05-14 by marvinroger