homieiot/homie-esp8266 #87: Wishlist for v2.0.0

marvinroger, 2016-05-14:
### TODO

* [ ] The framework must be completely asynchronous. It was not doable before because TCP stuff was synchronous, now with [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP), we can. The only drawback is SSL is not supported. This is currently not stable enough anyway, and it probably won't be production ready due to limited memory, something that might change with the ESP32.

* [ ] There must be custom settings support (#26)

* [ ] There must be a way to change configuration while in normal mode. This is achievable with [AsyncMqttClient](https://github.com/marvinroger/async-mqtt-client) because there is no hard limits like PubSubClient has (#36)

* [ ] Embed the magic bytes (useful for [homie-ota](https://github.com/jpmens/homie-ota) for example) as a macro. Something like:

```c++
#define Homie_setFirmware(name, version) const char* __FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" name "\x93\x44\x6b\xa7\x75"; const char* __FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" version "\xb0\x30\x48\xd4\x1a"; Homie.setFirmware(__FLAGGED_FW_NAME, __FLAGGED_FW_VERSION);
```

Should do, therefore `setFirmware` shoud only take data in-between the flags. As such, the current `setFirmware` (that will be renamed) will only be used by the macro, and must not be used in the normal sketch.

### Let's discuss

The 2.0.0 will be a breaking change. As such, we can safely change the API, the functions name, the order of the parameters... Let me know if you think there is any weirdness with the API, or the HTTP API of the config mode, etc. :smiley:

jpmens, 2016-05-14:
+1 (and very much appreciated) for embedding the magic bytes for homie-ota; this will make our sketches look neater. 

Regarding TLS (SSL): in spite of the huge amount of work you went to to get it to work, I have since reconsidered. Within a home, It's more important that devices are stable and "unbreakable" (eg killing them with PUBs) than having TLS. Furthermore I have the feeling that most people don't care about the security anyhow... In other words: drop TLS until ESP8266 supports it natively and asynchronously.

piotrwachowiak, 2016-05-14:
Hi, once again thanks for great work. If it is the right place for some kind of wish list:
 For me, if not TLS and/or native PKI support would be great, if only esp can handle this.
 Deep sleep and battery control included in MQTT convention, like delayedHomieLoop(int sleepTime) or something like this, method which takes sleepTime as argument and handles everything. Just dreaming.
 I had some troubles creating universal OLED thing, which is capable of displaying given topic/message with given params. I'd love to see it done in professional way. 
 It's probably just me, but event handling and subscribing is not clear for me - personally I failed with subscribe(), HomieNode relayNode("room", "relay", relayOnHandler, true) works, but I am not sure it's right way. 

Best regards,

Piotr

unaiur, 2016-05-14:
I wouldn't add custom settings. I think that we should allow to use WPS to connect to the WiFi network and use mDNS to discover a default mqtt broker service. Then, the remaining options can be configured with mqtt messages.

This way we didn't need any special BootConfig mode.

jpmens, 2016-05-14:
+1 for configuration via MQTT.

marvinroger, 2016-05-15:
@unaiur this would be ideal, but I am not sure it would fit everyone. WPS is probably not supported by all routers, and who actually advertise its MQTT broker via mDNS?

euphi, 2016-05-15:
Regarding API: Make the callback function a virtual method of HomieNode.

This would allow for a more modular design.

You still can have the default implementation to call a NodeInputHandler.

marvinroger, 2016-05-15:
@euphi Can you provide a sample code of how it would look like?

euphi, 2016-05-15:
Proof of concept:
[patch_virtual_callback.txt](https://github.com/marvinroger/homie-esp8266/files/265181/patch_virtual_callback.txt)

It was necessary to remove some const. Since I haven't coded for almost 6 years, I am a little bit out of practice, but I think the const was wrong there anyhow. With the call via the method pointer, you discard the const of the Homienode object - with the member function, this is no longer possible.
And it is quite obvious that calling a callback method that expect to change a state of a object, should not be called on a const object :-)

I did not had the time to create a class derived from Homienode, but this patch at least do not break anything :-)
Due to a business trip, I won't have time to try more for at least one week. :-(

euphi, 2016-05-16:
Due to bad weather today I found some time to make use of the new HomieNode::Inputhandler(..)

I created a new [project](https://github.com/euphi/BalkonController/
) at GitHub.

Especially have a look at https://github.com/euphi/BalkonController/blob/master/src/RGBWNode.cpp and 
https://github.com/euphi/BalkonController/blob/master/src/RGBWNode.h.

It is work in progress, there are still some nodes with the old style of callback.

To use them with OpenHab, I use the following items configuration:

    Number balkon_temp		"Temperatur [%.1f °C]"	(Balkon, TempSens)	{mqtt="<[local:devices/balkctrl/Temperatur/degrees:state:default]"}  
    Number balkon_press		"Luftdruck [%.1f hPa]"	(Balkon)			{mqtt="<[local:devices/balkctrl/Temperatur/hPa:state:default]"}  
    Switch balkon_auto		"Automatik" 			(Balkon)			{mqtt=">[local:devices/balkctrl/Auto/on/set:command:ON:true],>[local:devices/balkctrl/Auto/on/set:command:OFF:false]"}  
    Switch balkon_pumpe		"Pumpe" 				(Balkon)			{mqtt="<[local:devices/balkctrl/Pumpe/on:state:ON:true],<[local:devices/balkctrl/Pumpe/on:state:OFF:false],>[local:devices/balkctrl/Pumpe/on/set:command:ON:true],>[local:devices/balkctrl/Pumpe/on/set:command:OFF:false]"} 
    Switch balkon_valve1    "Kräuter bewässern (V1)"	(Balkon)		{mqtt="<[local:devices/balkctrl/Ventil/1:state:ON:true],<[local:devices/balkctrl/Ventil/1:state:OFF:false],>[local:devices/balkctrl/Ventil/1/set:command:ON:true],>[local:devices/balkctrl/Ventil/1/set:command:OFF:false]"}
    Switch balkon_valve2    "Hecke bewässern (V2)"	(Balkon)			{mqtt="<[local:devices/balkctrl/Ventil/2:state:ON:true],<[local:devices/balkctrl/Ventil/2:state:OFF:false],>[local:devices/balkctrl/Ventil/2/set:command:ON:true],>[local:devices/balkctrl/Ventil/2/set:command:OFF:false]"}
    Switch balkon_valve3    "Himbeeren bewässern (V3)"	(Balkon)		{mqtt="<[local:devices/balkctrl/Ventil/3:state:ON:true],<[local:devices/balkctrl/Ventil/3:state:OFF:false],>[local:devices/balkctrl/Ventil/3/set:command:ON:true],>[local:devices/balkctrl/Ventil/3/set:command:OFF:false]"}
    Switch balkon_valve4    "Rosen bewässern (V4)"	(Balkon)			{mqtt="<[local:devices/balkctrl/Ventil/4:state:ON:true],<[local:devices/balkctrl/Ventil/4:state:OFF:false],>[local:devices/balkctrl/Ventil/4/set:command:ON:true],>[local:devices/balkctrl/Ventil/4/set:command:OFF:false]"}
    Number balkon_signal	"WiFi Signal [%.1f %%]" (Balkon)			{mqtt="<[local:devices/balkctrl/$signal:state:default]"}

    Color  balkon_led		"LED-Streifen"	<slider>	(Balkon)
    Dimmer balkon_led_R		"Rot [%d %%]"	<switch>	(Balkon)		{mqtt=">[local:devices/balkctrl/LEDs/r/set:state:*:default]"}
    Dimmer balkon_led_G		"Grün [%d %%]"	<switch>	(Balkon)		{mqtt=">[local:devices/balkctrl/LEDs/g/set:state:*:default]"}
    Dimmer balkon_led_B		"Blau [%d %%]"	<switch>	(Balkon)		{mqtt=">[local:devices/balkctrl/LEDs/b/set:state:*:default]"}
    Dimmer balkon_led_W		"Weiß [%d %%]"	<slider>	(Balkon)		{mqtt=">[local:devices/balkctrl/LEDs/w/set:command:*:default]"}

unaiur, 2016-05-20:
I've evolved the proposal from @euphi in a project I'm doing to control led strips with Homie.
You can see the code in [homie-nodes](https://github.com/unaiur/homie-nodes)

TommySharpNZ, 2016-05-21:
To be able to run Homie on a battery powered device would be a great addition especially for simple environmental sensors around the house.

powerwade, 2016-05-23:
My wishlist item is to log/send messages to a local / remote syslog server.

jpmens, 2016-05-23:
@powerwade I think that would be out of scope, in particular as it'd mean implementing _syslog_ protocol: Homie is MQTT.

This can very easily be accomplished by creating a subscriber (server-side) which subscribes to your MQTT messages and pushes them out via syslog; a few lines of, say, Python would do the trick.

jayaras, 2016-05-23:
I could see syslog being handy and have thought of a few use cases where it would/could be.  At least for prototyping when I don't have serial output.   I've been fighting with a faulty pn532 board and it would have been handy to get some debugging info over syslog.

flaviostutz, 2016-05-24:
I like Homie API names and the code style very much. I made a feature request on https://github.com/marvinroger/homie-esp8266/issues/26 regarding to captive portal.

euphi, 2016-05-24:
Regarding syslog: It would be nice, if Homie could use another logger. Therefore it should be possible to set a logger class, e.g. any class implemting the "Print" class.

So, it is possible to create a syslog logger from another library and set it as preferred logger for Homie.

marvinroger, 2016-05-29:
Regarding the `syslog`: @jpmens stated deservedly that MQTT is a mean of communication. You could just create a `system` (or `log` or whatever you want) on which you publish your debugging messages to.

I do agree with @euphi however, being able to set a `Print` class to log to.

lighthousebulb, 2016-05-31:
I love the proposed Idea to set a configuration via mqtt.
But I have another suggestion:
What about a several configuration-files with different WiFi-settings for example? 
The Node should scan for available Networks and choose a configuration depending on the available Networks.
For example, when I set up a Node an move it to my friends house, with different WiFi-Settings, the Node could simply choose the right config-file for the different environment.

unaiur, 2016-06-01:
I'm working in wifi wps setup. I'm planning to add a new configuration option EnableWPS. When defined it will use the integrated settings in the ESP8266WiFi library, or start WPS pairing if no network is configured. It's nearly done and it uses very little code.

unaiur, 2016-06-01:
Another future ticket I would like to work at in the future is startup time.

I'm planning to implement a button on battery. When the button is pressed, the ESP will wake up from deep sleep, connect to the MQTT server and publish a topic.

The problem is that it needs about 4 seconds. I will try to:
a. Save DHCP lease to persistent memory and bypass this phase while the lease isn't stale.
b. Save mDNS service also.

unaiur, 2016-06-01:
You can see my WPS implementation at https://github.com/unaiur/homie-esp8266/tree/wps
It isn't fully tested yet, but you can taste it.

marvinroger, 2016-06-06:
@unaiur it seems we can cut down the time needed to connect to the Wi-Fi by not calling `WiFi.begin()` on each boot, but let the SDK store the Wi-Fi config itself. (see https://github.com/z2amiller/sensorboard/blob/b33da405795761efce0dfa02f53b6932c87958ee/PowerSaving.md)

marvinroger, 2016-06-06:
I am also considering removing the `OTA` mode and handle everything in the `normal` mode. MQTT would be used for the OTA, so OTA would not happen over HTTP anymore. This is now possible thanks to AsyncMqttClient and https://github.com/esp8266/Arduino/commit/2f933e20fa45f7c76347089f75974f557645fb03

@jpmens what do you think about it? `homie-ota` would need some change, but I think having only one protocol to handle everything (except for the HTTP API) is a step forward.

jpmens, 2016-06-06:
@marvinroger am I understanding correctly that the new binary firmware would be sent to Homie via an MQTT publish? If so, I'm all for it, and we can easily change `homie-ota` to do it.

marvinroger, 2016-06-06:
@jpmens Yes, this is what I mean. Let's go for it, then :)

marvinroger, 2016-06-18:
@unaiur just achieved to connect to the Wi-Fi network in 1s in the latest Git. :smile: 
The code needs esp8266/Arduino 2.3.0-rc2 to run and needs either a change in `Arduino.h` or https://github.com/esp8266/Arduino/pull/2084 to be fixed, so you might not want to test.

marvinroger, 2016-06-18:
OTA over MQTT works great! Thanks @me-no-dev for `Update.runAsync(true)`. :innocent:

me-no-dev, 2016-06-18:
you are welcome :)

gazally, 2016-06-26:
I really like what you've done here, it's all very well thought out. But I have a wishlist that might take you to 3.0! 😄

I'd like to use Homie on an ESP8266 that is a packet radio gateway to battery-powered sensor devices that periodically wake and send data and are capable of downloading firmware updates as well. The packet radios have far greater range than wifi and use less power, at the expense of bitrate.

This means I need "child" devices which use the parent's wifi config and MQTT connection but have their own name, node list, uptime, firmware etc. My handler callbacks would then communicate with the sensors via packet radio.

But then thinking about the architecture, wouldn't it be great if the code in the sensors could look just like the code in the gateway? The user sketch code in your API now doesn't know anything about wifi or MQTT and wouldn't have to change if data was being sent by packet radio instead. To accomplish this there would need to be an API layer in your library where one configuration and communication protocol could be substituted for another. And then if someone wanted to communicate with child devices using bluetooth instead, they could write code for the API.

But once you have done that, the gateway's upstream communication is no longer limited to wifi. Communication protocols could be written for devices that get their IP address from a cellular modem, or simply an ethernet cable. Or someone might want to build gateways between different packet radio networks, where local gateways communicate with very low-power devices at short range and send the data at higher power to a gateway device which has an MQTT connection. Many, many possibilities!

gazally, 2016-06-26:
I've now spent some time reading your code and I see that the upstream link over packet radio is best handled with a new library called homie-rfm69hw or some such. Which maybe I will have to write myself. 😄 But the concept of child devices that share the parent's upstream connection is still on my wishlist, and an API for downstream communication with the children that allows different methods of communication to be used transparently to the top-level code.

TommySharpNZ, 2016-06-27:
@Gazally, have you looked into something like MySensors for what you are thinking?

marvinroger, 2016-06-27:
@gazally this is indeed a good idea, but I am not sure this would be a good fit for this library. I'd like to keep Homie for ESP8266 (relatively) simple. It's meant to be used with MQTT. If ESP8266s were pricey, I would probably reconsider, but at ~3$ each, I think Wi-Fi is a great fit for IoT.

But, what you're asking is doable, even with the current codebase. You only have to mirror the MQTT communications, using the protocol you want. This approach is not as extensible as what you're asking for, but as @TommySharpNZ said, libraries like MySensors are great at handling that. :)

gazally, 2016-06-27:
Thanks for the suggestions! I'll have a look at MySensors.

Nzbuu, 2016-07-01:
In case you weren't aware, esp8266/Arduino 2.3.0 was released last week: https://github.com/esp8266/Arduino/releases

marvinroger, 2016-07-01:
@Nzbuu I was, thanks. :)
But there is a lot of work left before v2 is ready.

hschroyen, 2016-07-07:
Hi Marvin,

I am following your project very closely and like the idea of the framework.
I also stumbled upon the following project which might be very interesting to incorporate some idea's in homie.

http://harizanov.com/wiki/wiki-home/three-channel-wifi-relaythermostat-board/

Keep up the good work

marvinroger, 2016-08-19:
@hschroyen thanks, I'll take a look at it. :smiley: 

All new features are implemented. The next step will be to make sure it's bulletproof (you can easily crash it by sending big payload for the moment), then it will be documentation time, as a lot of things changed (for the good).

marvinroger, 2016-08-20:
Documentation is up: https://homie-esp8266.readme.io/ :smile: 
Let me know what you think. Please note it's not yet released, I want the v2 to be a minimum stable, without dumb negligences (though there will probably be some anyway)

euphi, 2016-08-21:
I proposed some edit yesterday (description of the `virtual handleInput(..)` as fourth input handler variant.

Unfortunately there is no feedback on readme.io, so I'm unsure if you received it?

Furthermore, I think an upgrade guide from 1.5.0 would be useful. I cannot propose a new page on readme.io, so I will list some suggestions here:

## Summary of important Changes
* Replace platformio libraries (PubSubClient replaced by AsyncMQTT)
* Important API changes
  * No need to use `registerNode(..)`
  * There is no more `subsribeToAll(..)`
  * Use `advertise(..)` to create properties
  * Settable properties must be marked as such by calling  `settable(..)`.
  * ....

## Caveeats
  * Signature of `setNodeProperty()` changed, but still compiles fine.

marvinroger, 2016-08-21:
I'm sorry, my Internet connection is rather limited until I come back from
vacations, so I won't work on the project for a week. Thanks for
contributing to the docs, I'll take a look. It would be a good idea to
create a new page about extending HomieNode, as it is a feature that
deserves it's own space.

jpmens, 2016-08-29:
@unaiur has there been any progress with WPS for homie-esp8266? I am in a situation where this would probably be extremely useful for us.

shogsbro, 2016-08-31:
Would it make sense to implement a captive portal while in Configure mode? I think it would only require capturing the iOS and Android test URL's, and returning a HTTP redirect.

marvinroger, 2016-08-31:
@shogsbro the captive portal is already implemented in the latest git, but this is only useful if you've embedded the `ui_bundle.gz`

thorhs, 2016-09-06:
I too would love a battery friendly deep sleep. Is that possible?  I have a custom code for temperature sensing using ESP and MQTT. I would love to swap out the plumbing for Homie.

marvinroger, 2016-09-07:
Deep sleep will be possible with #138, I don't know if there's something else you would like for deep sleep?

karlp, 2016-11-10:
Where's the documentation on the "ui_bundle.gz" ?  I can't find it in the 2.0 docs online.

marvinroger, 2016-11-14:
@karlp just added a page: https://homie-esp8266.readme.io/v2.0.0/docs/ui-bundle

flaviostutz, 2016-11-19:
@tiagostutz, see what an amazing v2.0 planning and execution for Homie 2.0...

tiagostutz, 2016-11-19:
Awesome! Great expectations here...!
Now let's rush with the Config JSON API Emulator project to see if it helps build the captive portal... When I finish the README.md I'll share here to get some insights from the community.

tiagostutz, 2016-11-19:
The first version of the emulator is ready. It's not a big deal, but maybe it can be useful:

* https://github.com/stutzlab/homie-esp8266-config-emulator
* https://hub.docker.com/r/stutzlab/homie-esp8266-config-emulator/

averri, 2016-11-22:
The only point I don't like for v2 is removing SSL. We can't have a IoT device exchanging data with Internet with no security.

jpmens, 2016-11-22:
@averri quite right, which is why you probably shouldn't be sending this kind of data over the Internet in the first place. Marvin has profusely explained why TLS is currently not possible. IMO you have three options:

1. Payload encryption (using whichever method you prefer). This, though, will not encrypt metadata such as topic names obviously, nor will it encrypt authentication credentials.
2. Satellite brokers. Use a broker which is (network-wise) very close to your "things". These connect via plain MQTT to the broker which then bridges out (via TLS or TLS-PSK) to upstream brokers. This is what we do and it's very reliable because the portion of the network where most hiccups occur is coupled by brokers which can elegantly retry/recover from outages
3. Use something else. (Though I'm not aware of any ESP8266 framework which comes close to what Homie offers in terms of functionality.)

boneskull, 2016-11-22:
> Marvin has profusely explained why TLS is currently not possible

Where is this?  I was trying to touch AWS's IoT service, but ended up having to use a proxy.

averri, 2016-11-22:
Hi @jpmens, I sincerely don't understand the rationale of having an IoT device which we can't use for sending data to the internet. So going through the options:

1. It's not possible to have partial encryption - or you have it or you don't have it. Neglecting security is a severe issue. See AWS IoT for example, it is very incipient, but it has strong security since the beginning.
2. Agree, this is one possible solution, but we are limiting the possibilities. What if someone wants the device to connect directly to a MQTT server on the internet? 
3. It's correct to say that actually there is no other framework that comes close to Homie regarding its features. Homie it is really great, I liked it since the first time I read about it. That's why I am here contributing with critical ideas. Everybody can use something else, but Homie will continue to be unsecure if you remove TLS.

I do understand the difficulties for adding TLS to Homie at the moment, but I would like to say: don't give up because it is complex. Maybe you may not add it to v2.0.0, but I strongly recommend Homie developers to consider it as the next big feature.

Bear in mind: security will be a decision factor for adoption by many people.

boneskull, 2016-11-26:
I have a use case that may already be covered, but I'm not sure of the canonical way to approach it.

I have a device controlling a strip of NeoPixels.  One function is that it will cycle through the colors (a [rainbow](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/strandtest/strandtest.ino#L60)).  The speed at which it cycles is currently configurable at a setting.

I could change the speed by exposing a property on a `HomieNode`.  I'd like to persist this speed between resets.  The only way to do this (to my knowledge) is to have both a custom setting and property handler for "speed"; I can have the property handler set the custom setting, but that seems like a smell.

Ideas:

- Allow user-defined handler for `$implementation/config/set` events.  I don't see this event type mentioned in [the docs](https://homie-esp8266.readme.io/docs/events).  In my case, could inspect the custom settings, then update how the "rainbow" task is working.  But I could see other users  changing WiFi and other stuff on-the-fly, or causing a reset, etc.
- Persist `HomieNode` property state.  I don't know what this would look like.

boneskull, 2016-11-26:
...I'm not sure if I *can* actually write a setting from my code other than to broadcast?

marvinroger, 2016-11-30:
@averri we don't give up, we will probably add it back in the future. 😉 
@boneskull updating the configuration through `$implementation/config/set` reboots the device anyway. Just send your `speed` property `retained`, and your device will receive it after a reboot, nope?

boneskull, 2016-11-30:
@marvinroger Good idea. I'm an MQTT novice.

euphi, 2017-01-13:
Is it already possible to use platformio for latest v2.0.0? It seems there is a `library.json` file for 2.0.0, but in the development branch, of course.

platformio added some versioning support recently, but I'm unsure how it works?
I tried to add "Homie@2.0.0" as dependency for a project, that did not work.

Note: I ask here first to get some explanation. If no one knows if it is possible and how, I will ask again at that platformio community.

benzino77, 2017-01-14:
Try this in platformio.ini:
```
...
lib_deps =
  https://github.com/marvinroger/homie-esp8266.git 
...
```

[4 more replies]

closed 2019-05-21 by stritti