homieiot/homie-esp8266 #93: Converting an existing sketch to work with Homie?

TommySharpNZ, 2016-05-21:
I've been struggling for a few days now to convert a sketch I found and make it work using the homie framework.

What I'm trying to achieve is counting the pulses on a power meter and sending each pulse and also a "watts" value to MQTT.

As far as I can tell this is a fully working sketch, maybe someone can help out?
https://jonarcher.info/2014/03/arduino-based-electricity-monitor/
https://github.com/jfarcher/energymonitor/blob/master/energymonitor.ino

euphi, 2016-05-21:
The sketch is > 80% network / mqtt code, so it is not needed with homie, because homie does all this stuff for you. 

Best way is to start with a homie example sketch for a sensor like the temperatureSensor example and replace the temperature related code with the power meter code. 
In setup(), you only need to copy the pinMode() and attachInterrupt().
Copy the whole loop() function to the loopHandler() and remove network code.

marvinroger, 2016-05-29:
I could not have said it better than @euphi did :wink: 
Any success @TommySharpNZ?

TommySharpNZ, 2016-05-29:
Hey guys, yeah @euphi and his instructions were VERY helpful. I have my power sensor up and running as of this morning. It was the "attachInterrupt" that was throwing me.

@marvinroger, I think the more working examples you have the better so I'll post up my sketch once I know it's working well.

marvinroger, 2016-06-02:
@TommySharpNZ great, post it here if / when you want to.

closed 2016-06-02 by marvinroger